chris1111/Wireless-USB-Big-Sur-Adapter #1: Failed when "validating the software".

lulujyc, 2020-06-26:
Log:

Jun 26 20:59:52 10 installd[678]: PackageKit: Client requested install as user consented; however client PKInstallDaemonClient pid=1236, uid=501 (/System/Library/CoreServices/Installer.app/Contents/MacOS/Installer) is not entitled. User consent request denied.
Jun 26 20:59:52 10 installd[678]: PackageKit: ----- Begin install -----
Jun 26 21:00:06 10 installd[678]: PackageKit: Install Failed: Error Domain=PKInstallErrorDomain Code=112 "运行软件包“Wireless USB Big Sur AdapterV-1.pkg”的脚本时出错。" UserInfo={NSFilePath=./postinstall, NSURL=file:///var/folders/dl/dy0ppnr128jdnrlypy5xpghc0000gn/T/wzQL.VBVrmf/Wireless%20USB%20Big%20Sur%20AdapterV-1.pkg#drivers.pkg, PKInstallPackageIdentifier=com.109Driver.wirelessUsbBigsuradapterv1.Drivers.pkg, NSLocalizedDescription=运行软件包“Wireless USB Big Sur AdapterV-1.pkg”的脚本时出错。} {
	    NSFilePath = "./postinstall";
	    NSLocalizedDescription = "\U8fd0\U884c\U8f6f\U4ef6\U5305\U201cWireless USB Big Sur AdapterV-1.pkg\U201d\U7684\U811a\U672c\U65f6\U51fa\U9519\U3002";
	    NSURL = "file:///var/folders/dl/dy0ppnr128jdnrlypy5xpghc0000gn/T/wzQL.VBVrmf/Wireless%20USB%20Big%20Sur%20AdapterV-1.pkg#drivers.pkg";
	    PKInstallPackageIdentifier = "com.109Driver.wirelessUsbBigsuradapterv1.Drivers.pkg";
	}
Jun 26 21:00:07 10 Installer[1236]: Install failed: 安装器遇到了一个错误，导致安装失败。请联系软件生产企业以获得帮助。
Jun 26 21:00:07 10 Installer[1236]: Displaying 'Install Failed' UI.
Jun 26 21:00:07 10 Installer[1236]: 'Install Failed' UI displayed message:'安装器遇到了一个错误，导致安装失败。请联系软件生产企业以获得帮助。'.

OS: macOS 11 DP1

Hardware: AMD 1950X via Vanilla OpenCore

chris1111, 2020-06-26:
Provide a full log then not in Chinese please
Thanks

lulujyc, 2020-06-26:
> Provide a full log then not in Chinese please
> Thanks

Sorry I can't change the log output language but the translations are:
Jun 26 21:00:06 10 installd[678]: PackageKit: Install Failed: Error Domain=PKInstallErrorDomain Code=112 "运行软件包“Wireless USB Big Sur AdapterV-1.pkg”的脚本时出错。" UserInfo={NSFilePath=./postinstall, NSURL=file:///var/folders/dl/dy0ppnr128jdnrlypy5xpghc0000gn/T/wzQL.VBVrmf/Wireless%20USB%20Big%20Sur%20AdapterV-1.pkg#drivers.pkg, PKInstallPackageIdentifier=com.109Driver.wirelessUsbBigsuradapterv1.Drivers.pkg, NSLocalizedDescription=运行软件包“Wireless USB Big Sur AdapterV-1.pkg”的脚本时出错。} {

Jun 26 21:00:06 10 installd[678]: PackageKit: Install Failed: Error Domain=PKInstallErrorDomain Code=112 "Running the script of the package “Wireless USB Big Sur AdapterV-1.pkg” encountered errors." UserInfo={NSFilePath=./postinstall, NSURL=file:///var/folders/dl/dy0ppnr128jdnrlypy5xpghc0000gn/T/wzQL.VBVrmf/Wireless%20USB%20Big%20Sur%20AdapterV-1.pkg#drivers.pkg, PKInstallPackageIdentifier=com.109Driver.wirelessUsbBigsuradapterv1.Drivers.pkg, NSLocalizedDescription=Running the script of the package “Wireless USB Big Sur AdapterV-1.pkg” encountered errors.} {

and

 Jun 26 21:00:07 10 Installer[1236]: Install failed: 安装器遇到了一个错误，导致安装失败。请联系软件生产企业以获得帮助。
Jun 26 21:00:07 10 Installer[1236]: Displaying 'Install Failed' UI.
Jun 26 21:00:07 10 Installer[1236]: 'Install Failed' UI displayed message:'安装器遇到了一个错误，导致安装失败。请联系软件生产企业以获得帮助。'.

 Jun 26 21:00:07 10 Installer[1236]: Install failed: Package installer encountered an error which caused the installation to fail. Please contact the software provider.
Jun 26 21:00:07 10 Installer[1236]: Displaying 'Install Failed' UI.
Jun 26 21:00:07 10 Installer[1236]: 'Install Failed' UI displayed message:'Package installer encountered an error which caused the installation to fail. Please contact the software provider.'.

chris1111, 2020-06-26:
I see lots people have problems with a WIFI USB adapter and Open Core as boot loader
I don't know why because I not using Open core
Then you have also a AMD patching in a Beta phase for BS so I can not rely help here sorry

closed 2020-06-26 by chris1111