chris1111/Wireless-USB-Big-Sur-Adapter #2: Failed Validating the Software

jvaglivelo, 2020-06-27:
Failing to validate the software on Big Sur.

Log: 

> 2020-06-26 23:20:17-04 admin installd[601]: PackageKit: releasing backupd

> 2020-06-26 23:20:17-04 admin installd[601]: PackageKit: allow user idle system sleep

> 2020-06-26 23:20:17-04 admin installd[601]: PackageKit: Cleared responsibility for install from 1891.

> 2020-06-26 23:20:17-04 admin installd[601]: PackageKit: Cleared permissions on Installer.app

> 2020-06-26 23:20:17-04 admin installd[601]: PackageKit: Install Failed: Error Domain=PKInstallErrorDomain Code=112 "An error occurred while running scripts from the package “Wireless USB Big Sur AdapterV-1”." UserInfo={NSFilePath=./postinstall, NSURL=file:///Volumes/Install%20macOS%20Beta/Wireless%20USB%20Big%20Sur%20AdapterV-1.pkg#drivers.pkg, PKInstallPackageIdentifier=com.109Driver.wirelessUsbBigsuradapterv1.Drivers.pkg, NSLocalizedDescription=An error occurred while running scripts from the package “Wireless USB Big Sur AdapterV-1”.} {
> 	    NSFilePath = "./postinstall";
> 	    NSLocalizedDescription = "An error occurred while running scripts from the package \U201cWireless USB Big Sur AdapterV-1\U201d.";
> 	    NSURL = "file:///Volumes/Install%20macOS%20Beta/Wireless%20USB%20Big%20Sur%20AdapterV-1.pkg#drivers.pkg";
> 	    PKInstallPackageIdentifier = "com.109Driver.wirelessUsbBigsuradapterv1.Drivers.pkg";
> 	}

> 2020-06-26 23:20:17-04 admin installd[601]: PackageKit: Running idle tasks

> 2020-06-26 23:20:17-04 admin Installer[1891]: install:didFailWithError:Error Domain=PKInstallErrorDomain Code=112 "An error occurred while running scripts from the package “Wireless USB Big Sur AdapterV-1”." UserInfo={NSFilePath=./postinstall, NSURL=file:///Volumes/Install%20macOS%20Beta/Wireless%20USB%20Big%20Sur%20AdapterV-1.pkg#drivers.pkg, PKInstallPackageIdentifier=com.109Driver.wirelessUsbBigsuradapterv1.Drivers.pkg, NSLocalizedDescription=An error occurred while running scripts from the package “Wireless USB Big Sur AdapterV-1”.}

> 2020-06-26 23:20:17-04 admin installd[601]: PackageKit: Removing client PKInstallDaemonClient pid=1891, uid=502 (/System/Library/CoreServices/Installer.app/Contents/MacOS/Installer)

> 2020-06-26 23:20:17-04 admin installd[601]: PackageKit: Done with sandbox removals

> 2020-06-26 23:20:17-04 admin Installer[1891]: Install failed: The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance.

> 2020-06-26 23:20:17-04 admin Installer[1891]: IFDInstallController 2DB0AC0 state = 8

> 2020-06-26 23:20:17-04 admin Installer[1891]: Displaying 'Install Failed' UI.

> 2020-06-26 23:20:17-04 admin Installer[1891]: 'Install Failed' UI displayed message:'The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance.'.
> 

System:
i7-6700k
OpenCore 0.6.0
macOS 11.0 Beta

chris1111, 2020-06-27:
Not working in Open Core and AMD

closed 2020-06-27 by chris1111

jvaglivelo, 2020-06-27:
I'm not using AMD