chris1111/Wireless-USB-Big-Sur-Adapter #6: Not working on MacBook Pro (Retina, 13-inch, Late 2013)

4shadowr, 2020-07-15:
I followed your instructions from video
I get Installed Successfully, I restart
After restart the status bar icon says 
> No USB-WiFi Plugin
where as I have the adapter plugged in and my usb slots are also working
Tried both Releases v1 & v2
[Installer Log 15-Jul-2020 v2.txt](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/4924365/Installer.Log.15-Jul-2020.v2.txt)

System:- 
Device: MacBook Pro (Retina, 13-inch, Late 2013)
OS version: 11.0 Beta (20A4300b)
Adapter: TL-WN725N

chris1111, 2020-07-15:
What is the TL-WN725N ? If you have V1 and V2 no macOS driver for this adapter
https://www.tp-link.com/ca/support/download/tl-wn725n/
If you have the V3 that suppose to work if you have allow software from security ?
So uninstall then retry if you have V3 then let me know

4shadowr, 2020-07-15:
Its V3 and Yes, I have allowed it from 
> System Preferences -> Security & Privacy -> Allow Realtek

chris1111, 2020-07-15:
> Its V3 and Yes, I have allowed it from
> 
> > System Preferences -> Security & Privacy -> Allow Realtek
> 
> It was working with [Wireless USB Adapter Clover-V15](https://github.com/chris1111/Wireless-USB-Adapter-Clover/releases/tag/V15)

I have the same MacBook of you and its work here 
https://support.apple.com/kb/sp691?locale=fr_CA

4shadowr, 2020-07-15:
It's Not working on My MBP,
I have uploaded the installer file if that helps
Any Idea on how to troubleshoot this?

4shadowr, 2020-07-15:
Which beta are you running?
I am on Latest Beta (11.0 Beta (20A4300b))
Have you tried it on that ?

chris1111, 2020-07-15:
When you have Allow soft did you see multiple Soft trying to allow like LaCie or other things ?

4shadowr, 2020-07-15:
No, I Don't think so I just Clicked Allow and Restart

chris1111, 2020-07-15:
you don't have a native wifi on your MacBook

4shadowr, 2020-07-15:
I Uninstalled using the installer File -> Restarted
Re-installed it
This time it didn't ask me to allow
[image]
Restarted again too still not working
How to do a clean Re-install ?

4shadowr, 2020-07-15:
> you don't have a native wifi on your MacBook

I think Built-in WiFi Adapter is broken or something
No Stores are open due to pandemic :(

chris1111, 2020-07-15:
Strange because we don't have a installer

4shadowr, 2020-07-15:
> Strange because we don't have a Uninstaller file

NOT an Uninstaller file, the installer package
[image]

4shadowr, 2020-07-15:
Can you tell me with how to do a clean Uninstall
So I can do a Fresh Install again ?

chris1111, 2020-07-15:
Send me system report of your MacBook Pro on **About this Mac** ➤ /File /save 
then put here on ZIP file of the system report

4shadowr, 2020-07-15:
[MacBook Pro.zip](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/4925443/MacBook.Pro.zip)

chris1111, 2020-07-15:
boot to the recovery then try to setup SIP disable just for trying again
in the recovery open Utility /Terminal type
csrutil disable
Closed then reboot and retry the program
let me know

4shadowr, 2020-07-15:
Worked like a charm
Thanks !!

> boot to the recovery then try to setup SIP disable just for trying again
> in the recovery open Utility /Terminal type
> csrutil disable
> Closed then reboot and retry the program
> let me know

closed 2020-07-15 by 4shadowr