chris1111/Wireless-USB-Big-Sur-Adapter #9: Adaptor package failing to install on iMac 13,2

Calibre5, 2020-07-24:
Hi Chris, I've got an issue with the package failing to load on Big Sur beta 3. Gatekeeper has been disabled and running programs stopped but it fails after installing the kexts in L/E.
Here's the error log from the installer.
Thanks.
[Installer Log 24-Jul-2020.txt](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/4973504/Installer.Log.24-Jul-2020.txt)

Calibre5, 2020-07-24:
An update. The installer seems to fail when attempting to install Wlan.Software.plist in Library/LaunchAgents.
the file doesn't get its .plist extension.
Kexts and Application Support stuff get installed but nothing in LaunchDaemons.
I've added .txt extension so I could upload the file
[Wlan.Software.txt](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/4974181/Wlan.Software.txt)

chris1111, 2020-07-24:
Ok I see the issue
Release is suspend; need fix for Beta 3 so uninstall the program I will see what is not working

closed 2020-07-24 by chris1111

chris1111, 2020-07-24:
Update 24 July 2020 Fix for Beta3
- To using release V3 SIP Security must be disable

Its work 
[image]

Calibre5, 2020-07-26:
I ran the installer package v3 and it went through successfully. Everything installed OK in L/E, Application Support and LaunchAgents & launchDaemons. Also checked in Library/StagedExtensions/Library/Extensions and the 3 kexts are present.
Problems arise when I go to System Preferences/Security & Privacy/General to allow the Realtek drivers, there's no allow button. At this point I've got the app in the menu bar with a red dot and clicking it show's "No USB-WiFi  Plug in.
Gatekeeper is disabled as is SIP (via Recovery and terminal commands csrutil disable followed by Csrutil authenticated-root disable). Both commands returned confirmation in the terminal and after rebooting terminal in the OS shows SIP disabled.
I'm using a Linksys WUSB6300 with Product ID: 003F and Vendor ID:13b1, which match the settings in the kext plist.
Any ideas gratefully received. Thanks
[Installer Log 26-Jul-2020.txt](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/4977486/Installer.Log.26-Jul-2020.txt)

chris1111, 2020-07-26:
Ive made another release today
Ive just test again and I have Allow Realteak
[image]

On other system Hackintosh laptop I am not have allow but after reboot all is ok

Calibre5, 2020-07-26:
I've just downloaded the new version 3 and uninstalled the previous installation. Rebooted and checked Gatekeeper and SIP disabled. Ran the new installer and got "installation successful" message but still no Allow button showing. Menu app still showing  "No USB-wifi". No luck I'm afraid.
[image]

chris1111, 2020-07-26:
I don't know whats  wrong for you ? I ve test to 4 Machine this morning all is OK

Calibre5, 2020-07-26:
Forgot log.
[Installer Log 26-Jul-2020.txt](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/4977680/Installer.Log.26-Jul-2020.txt)

Calibre5, 2020-07-26:
OK, thanks for investigating. I'll go through the installation log and see if I can pinpoint anything.

chris1111, 2020-07-26:
macOS Big Sur 11 Beta 3 test on MacBook Pro late 2013
See Video
[image]

Calibre5, 2020-07-26:
Realised I'm not getting the message re. System Extension Updated as per your video. Probably why I don't get the Allow button. Is there a way to check which extensions have been allowed?
[image]

chris1111, 2020-07-26:
message suppose to be there

Calibre5, 2020-07-26:
Yes, but I'm not seeing it, even though I get a successful installation message at the end.

Calibre5, 2020-07-26:
Update. Just tried Big Sur beta 3 reinstallation from 12 GB package. No difference.
Ah well. Roll on beta 4!