chris1111/Wireless-USB-Big-Sur-Adapter #10: NO USB Wifi Device Plugged in

tlefko, 2020-07-25:
Followed all instructions for gatekeeper and SIP to a tee. The device is detected in system information but the menu bar shows no USB wifi detected. On beta 20A4300b. Know it's not the USB wifi as works fine under windows and worked under Catalina

chris1111, 2020-07-25:
Uninstall 
reboot to recovery Enable SIP reboot then reboot to recovery again and Disable SIP
Run the program Allow Realtek only, uncheck Other Allow

tlefko, 2020-07-25:
will try and update results @chris1111

chris1111, 2020-07-25:
Its work here

closed 2020-07-25 by chris1111

chris1111, 2020-07-26:
macOS Big Sur 11 Beta 3 test on MacBook Pro late 2013
See Video
[image]

tlefko, 2020-07-26:
> macOS Big Sur 11 Beta 3 test on MacBook Pro late 2013
> See Video
> [image]

yeah that happened to me but still no USB found. I will try on laptop now

tlefko, 2020-08-03:
@chris1111 marked as resolved -- make sure you instruct users to have sip completely disabled not just thru apple internal --- CSR (Cambridge silicon) needs to be disabled in extensions

chris1111, 2020-08-03:
> @chris1111 marked as resolved -- make sure you instruct users to have sip completely disabled not just thru apple internal --- CSR (Cambridge silicon) needs to be disabled in extensions
this is written Working SIP disable on Beta 3

tlefko, 2020-08-03:
> > @chris1111 marked as resolved -- make sure you instruct users to have sip completely disabled not just thru apple internal --- CSR (Cambridge silicon) needs to be disabled in extensions
> 
> this is written Working SIP disable on Beta 3

I ran on Beta 2 -- I had sip disabled through recovery which returns Sip disabled (apple Internal). this won't work. Enabling CSR breaks Wifi as well

chris1111, 2020-08-03:
of course that yes if you have allocated extensions before it can cause problems you must understand that it is a BETA and only the third

tlefko, 2020-09-30:
> of course that yes if you have allocated extensions before it can cause problems you must understand that it is a BETA and only the third

yeah you're totally right. I didn't realize because the installer tries to have you allow caimbridge but that's my model of USB, maybe include in the instructions to do Realtek only. great work

jeremybeny, 2020-11-16:
> Followed all instructions for gatekeeper and SIP to a tee. The device is detected in system information but the menu bar shows no USB wifi detected. On beta 20A4300b. Know it's not the USB wifi as works fine under windows and worked under Catalina.
> 
> **Note at first the prompt for extension showed up, allowed it. After uninstall and reinstall it no longer appears. -- in both cases no USB found. My csrutil status is disabled (apple internal)
> 
> Any suggestions? The USB is supported and is realtek (exact model not known but worked under catalina) something AC1200M
> Tried both USB 3 and USB 2. Note* No options were added to network preferences. Tried reset them no luck

can you help me solving the same problem? I have done everything posted here but still not detecting my usb Edimax EW-7822ULC  thank you