chris1111/Wireless-USB-Big-Sur-Adapter #11: Just got public beta for Big Sur, the driver for netgear 7000 adapter is not working

jclocker, 2020-08-06:
The new driver for Big Sur is not working with the netgear 7000 adapter getting no usb-wifi plug in and no blue light on the A7000 adapter connected to USB port and tested with USB powered hub. Not seeing the adapter on MacBook Pro 16 2009 model.
[image]

chris1111, 2020-08-06:
Do you have SIP security turned off? Did you also follow all the steps?

jclocker, 2020-08-06:
Chris will have to check. Will let you know.

closed 2020-08-27 by chris1111

jclocker, 2020-09-03:
Chris in the video you show running gatekeeper.command from the desktop folder. Is this a script? Still having problems with the Big Sur showing No USB-WIFI  Plug in. The Netgear A7000 works when I am in bootcamp and window 10 but still not in Big Sur. 

Thanks for any help Jim a fan of your software