chris1111/Wireless-USB-Big-Sur-Adapter #13: Support for T2U Plus please

deveraux03, 2020-08-12:
T2U Plus doesn't work. Can you please add support for this model? Thank you!

chris1111, 2020-08-12:
Its work you need SIP security completely Disable

closed 2020-08-12 by chris1111

deveraux03, 2020-08-12:
i’ve disabled that. the adapter won’t even light up. 

i can try re installing. how do i completely remove everything and install again ?

deveraux03, 2020-08-12:
How to clean install again? When I install again "allow" in system preferences does not show up

chris1111, 2020-08-12:
Uninstall the program  Boot in Recovery Enable SIP reboot on System then  reboot on recovery Disable SIP again then reinstall the program

deveraux03, 2020-08-12:
Tried that. Still doesn't ask for "allow".

chris1111, 2020-08-12:
> Tried that. Still doesn't ask for "allow".

Uninstall the program, Enable SIP then install the program reboot disable SIP then reinstall the program?

deveraux03, 2020-08-12:
Tried that just now. Still doesn't work :( 

Here's the log

Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: @(#)PROGRAM:Install  PROJECT:Install-1000
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: @(#)PROGRAM:Installer  PROJECT:Installer-1015.3
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Hardware: MacBookPro12,1 @ 2.70 GHz (x 4), 8192 MB RAM
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Running OS Build: macOS 11.0 (20A5343i)
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Env: USER=muathaz
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Env: __CFBundleIdentifier=com.apple.installer
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Env: COMMAND_MODE=unix2003
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Env: LOGNAME=muathaz
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Env: PATH=/usr/bin:/bin:/usr/sbin:/sbin
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Env: SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.QxH9XZgNEy/Listeners
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Env: SHELL=/bin/zsh
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Env: HOME=/Users/muathaz
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Env: __CF_USER_TEXT_ENCODING=0x1F5:0:2
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Env: TMPDIR=/var/folders/dt/pdvj_ggs53l5c10mmsnfs1v80000gn/T/
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Env: XPC_SERVICE_NAME=application.com.apple.installer.1152921500312158071.1152921500312158076
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Env: XPC_FLAGS=0x0
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Wireless USB BigSurAdapter-V-3  Installation Log
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Opened from: /Users/muathaz/Downloads/Wireless USB Big Sur AdapterV-3.pkg
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Package Authoring Error: <background> has an unsupported MIME type: image/data
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Package Authoring Error: <background_scaling> has an unsupported MIME type: X-NSObject/NSNumber
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Package Authoring Error: <background_alignment> has an unsupported MIME type: X-NSObject/NSNumber
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Package Authoring Error: <layout-direction> has an unsupported MIME type: X-NSObject/NSNumber
Aug 12 08:33:22 Muathazs-MacBook-Pro Installer[483]: Product archive /Users/muathaz/Downloads/Wireless USB Big Sur AdapterV-3.pkg trustLevel=100
Aug 12 08:33:23 Muathazs-MacBook-Pro Installer[483]: External component packages (5) trustLevel=100 (trust evaluation failed: Error Domain=PKInstallErrorDomain Code=101 "The package “Wireless USB Big Sur AdapterV-3” is not signed." UserInfo={NSURL=#uninstall.pkg -- file:///Users/muathaz/Downloads/Wireless%20USB%20Big%20Sur%20AdapterV-3.pkg#Distribution, PKInstallPackageIdentifier=com.109Driver.wirelessUsbBigsuradapterv3.uninstall.pkg, NSLocalizedDescription=The package “Wireless USB Big Sur AdapterV-3” is not signed.})
Aug 12 08:33:23 Muathazs-MacBook-Pro Installer[483]: Could not load resource license: (null)
Aug 12 08:33:23 Muathazs-MacBook-Pro Installer[483]: Package Authoring Error: Infinite loop between <choice> start_* attributes and mixed state aborted.
Aug 12 08:33:23 Muathazs-MacBook-Pro Installer[483]: Package Authoring Error: Infinite loop between <choice> attributes and mixed state aborted.
Aug 12 08:33:25 Muathazs-MacBook-Pro Installer[483]: Package Authoring Error: Infinite loop between <choice> start_* attributes and mixed state aborted.
Aug 12 08:33:25 Muathazs-MacBook-Pro Installer[483]: Package Authoring Error: Infinite loop between <choice> attributes and mixed state aborted.
Aug 12 08:33:26 Muathazs-MacBook-Pro SoftwareUpdateNotificationManager[508]: AssertionMgr: Take com.apple.softwareupdate.NotifyAgentAssertion-LoginDone assertion with type BackgroundTask for pid 508, id 0x812E
Aug 12 08:33:26 Muathazs-MacBook-Pro SoftwareUpdateNotificationManager[508]: SUOSUAuthenticationManager: Disabling local authentication requirementß
Aug 12 08:33:26 Muathazs-MacBook-Pro softwareupdated[509]: softwareupdated: Starting with build 11.0 (20A5343i)
Aug 12 08:33:26 Muathazs-MacBook-Pro softwareupdated[509]: Initializing SoftwareUpdateMacController (SUMacControllerServer)
Aug 12 08:33:26 Muathazs-MacBook-Pro softwareupdated[509]: Initialized SoftwareUpdateMacController (SUMacControllerServer)
Aug 12 08:33:26 Muathazs-MacBook-Pro softwareupdated[509]: SUOSUAlarmObserver: Setting alarm event stream handler
Aug 12 08:33:26 Muathazs-MacBook-Pro softwareupdated[509]: SUOSUAuthenticationManager: Disabling local authentication requirementß
Aug 12 08:33:26 Muathazs-MacBook-Pro softwareupdated[509]: SUOSUServiceDaemon: Error reading /var/folders/zz/zyxvpxvq6csfxvn_n00000s0000068/C/softwareupdated/com.apple.OSUpdate.SUOSUServiceDaemon.state: Error Domain=NSCocoaErrorDomain Code=260 "The file “com.apple.OSUpdate.SUOSUServiceDaemon.state” couldn’t be opened because there is no such file." UserInfo={NSFilePath=/var/folders/zz/zyxvpxvq6csfxvn_n00000s0000068/C/softwareupdated/com.apple.OSUpdate.SUOSUServiceDaemon.state, NSUnderlyingError=0x7f8b2ec28dc0 {Error Domain=NSPOSIXErrorDomain Code=2 "No such file or directory"}}
Aug 12 08:33:26 Muathazs-MacBook-Pro softwareupdated[509]: SUOSUServiceDaemon: Use standard 6 hour auto update interval
[... cut ...]
Aug 12 08:34:33 Muathazs-MacBook-Pro Installer[483]:      script      0.00 seconds
Aug 12 08:34:33 Muathazs-MacBook-Pro Installer[483]:        zero      0.01 seconds
Aug 12 08:34:33 Muathazs-MacBook-Pro Installer[483]:     install      49.34 seconds
Aug 12 08:34:33 Muathazs-MacBook-Pro Installer[483]:     -total-      49.39 seconds
Aug 12 08:34:33 Muathazs-MacBook-Pro Installer[483]: 
Aug 12 08:34:33 Muathazs-MacBook-Pro Installer[483]: IFDInstallController 3AD7800 state = 5
Aug 12 08:34:33 Muathazs-MacBook-Pro Installer[483]: Displaying 'Install Succeeded' UI.