chris1111/Wireless-USB-Big-Sur-Adapter #14: Kext files need to be rebuilt for Beta 5

CharlieWhile13, 2020-08-27:
When trying to load the kext file RtWlanDisk.kext, the error it spews out is: 
```
Executing: /usr/bin/kmutil load --bundle-path /Library/Extensions/RtWlanDisk.kext
Error: Error Domain=KMErrorDomain Code=1 "Error occurred while building cache: 
1: One or more binaries has an error which prevented linking.  See other errors.
2: Could not use 'com.realtek.driver.RtWlanDisk' because: Cannot find symbol for metaclass pointed to by '__ZN10RtWlanDisk10superClassE'.  Expected symbol '__ZN28IOSCSIPeripheralDeviceType0010gMetaClassE' to be defined in another kext
com.realtek.driver.RtWlanDisk specific: 
1: Cannot find symbol for metaclass pointed to by '__ZN10RtWlanDisk10superClassE'.  Expected symbol '__ZN28IOSCSIPeripheralDeviceType0010gMetaClassE' to be defined in another kext
" UserInfo={NSLocalizedDescription=Error occurred while building cache: 
1: One or more binaries has an error which prevented linking.  See other errors.
2: Could not use 'com.realtek.driver.RtWlanDisk' because: Cannot find symbol for metaclass pointed to by '__ZN10RtWlanDisk10superClassE'.  Expected symbol '__ZN28IOSCSIPeripheralDeviceType0010gMetaClassE' to be defined in another kext
com.realtek.driver.RtWlanDisk specific: 
1: Cannot find symbol for metaclass pointed to by '__ZN10RtWlanDisk10superClassE'.  Expected symbol '__ZN28IOSCSIPeripheralDeviceType0010gMetaClassE' to be defined in another kext
}
```
I'm guessing it's because it needs to be re-built for beta 5, going off of the change logs

chris1111, 2020-08-27:
which mac do you use?

CharlieWhile13, 2020-08-27:
MacBook Air 2015, Big Sur Developer Beta 5. I got that error from running `sudo kextutil /Library/Extensions/RtWlanDisk.kext`

chris1111, 2020-08-27:
Why runing kextutil

CharlieWhile13, 2020-08-27:
To try and figure out why it couldn't detect my WiFi card. With beta 5 any kext's need to be rebuilt using the new SDK, otherwise they won't work

chris1111, 2020-08-27:
did you try uninstall + reboot then Install again + reboot ?

CharlieWhile13, 2020-08-27:
Yes sir 

Xcode 12 beta 5 change notes:

```

Symbols exported by both xnu and Apple kernel extensions will change. Recompile your kernel extension with each update to macOS Big Sur 11 beta to ensure compatibility. (64262563)
```

chris1111, 2020-08-27:
Xcode and kernel extensions have nothing to do with this, extensions are supported
I just install again for a test and its work 
[image]
[image]

[image]

[image]

CharlieWhile13, 2020-08-27:
I disabled sip from recovery 
Ran the installer package 
Pressed allow in Settings 
Rebooted the computer 

Connected my card, and it said No USB WiFi Plugin

CharlieWhile13, 2020-08-27:
When checking every loaded kext file, nothing related to Realtek was there

chris1111, 2020-08-27:
I have the same message probably this is a bug from kmutil
run  kextstat

CharlieWhile13, 2020-08-27:
```
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
    1  127 0                  0          0          com.apple.kpi.bsd (20.0.0) F29B97A1-72E4-3D36-863F-8ADCCB731F78 <>
    2   11 0                  0          0          com.apple.kpi.dsep (20.0.0) F29B97A1-72E4-3D36-863F-8ADCCB731F78 <>
    3  163 0                  0          0          com.apple.kpi.iokit (20.0.0) F29B97A1-72E4-3D36-863F-8ADCCB731F78 <>
    4    0 0                  0          0          com.apple.kpi.kasan (20.0.0) F29B97A1-72E4-3D36-863F-8ADCCB731F78 <>
    5  168 0                  0          0          com.apple.kpi.libkern (20.0.0) F29B97A1-72E4-3D36-863F-8ADCCB731F78 <>
    6  147 0                  0          0          com.apple.kpi.mach (20.0.0) F29B97A1-72E4-3D36-863F-8ADCCB731F78 <>
    7   87 0                  0          0          com.apple.kpi.private (20.0.0) F29B97A1-72E4-3D36-863F-8ADCCB731F78 <>
    8   89 0                  0          0          com.apple.kpi.unsupported (20.0.0) F29B97A1-72E4-3D36-863F-8ADCCB731F78 <>
    9    2 0xffffff800316a000 0xe000     0xe000     com.apple.kec.Libm (1) 7D9632E1-D0E3-347D-A182-003C33041E59 <5>
   10   10 0xffffff800336d000 0x8e000    0x8e000    com.apple.kec.corecrypto (1.0) 7F6C6F3C-9901-3405-85B1-3FE4C0983C46 <8 7 6 5 3 1>
   11    0 0xffffff8003402000 0x8000     0x8000     com.apple.kec.pthread (1) 8C79821A-EA84-336F-9E85-09A687CDEE2C <8 7 6 5 3 1>
   12    1 0xffffff80034c3000 0x2000     0x2000     com.apple.driver.watchdog (1) 4812E55D-BEB7-3379-8C4A-38B6B47DE9B7 <8 7 6 5 3 1>
   13   36 0xffffff800267c000 0x2000     0x2000     com.apple.iokit.IOACPIFamily (1.4) 3EC07BB1-A5FF-387B-BBA8-C22A91181C4A <8 7 5 3>
   14   42 0xffffff8002bb8000 0x27000    0x27000    com.apple.iokit.IOPCIFamily (2.9) AF2A75C7-B656-33CA-A6C3-C6210D84E2EC <8 7 6 5 3>
   15   13 0xffffff800185f000 0x19000    0x19000    com.apple.driver.AppleSMC (3.1.9) A1F67D09-5AE8-376C-A662-6C2F2415455C <14 13 12 8 7 6 5 3 1>
   16    2 0xffffff800126e000 0x77000    0x77000    com.apple.driver.AppleACPIPlatform (6.1) 2407428B-FF05-3822-8F2D-8C054E3626C9 <15 14 13 8 7 6 5 3 1>
   17    5 0xffffff8002cbb000 0x2000     0x2000     com.apple.driver.IOSlaveProcessor (1) CFE500D5-D773-374D-9A20-B2D62E9A23CC <5 3>
   18    3 0xffffff8001845000 0x10000    0x10000    com.apple.driver.AppleSEPManager (1.0.1) 3768AC8C-1E8A-3C49-9811-63A8811336F5 <17 14 10 8 7 6 5 3 1>
   19    1 0xffffff800155c000 0x4000     0x4000     com.apple.driver.AppleBusPowerController (1.0) 6D135244-E6DD-3B33-BD40-69008CFF3582 <15 13 7 6 5 3>
   20    9 0xffffff80019a7000 0x4000     0x4000     com.apple.driver.usb.AppleUSBCommon (1.0) 4D575CB7-DB8A-3D22-9357-D114AA7310E1 <6 5 3 1>
   21    3 0xffffff8002fd4000 0x1000     0x1000     com.apple.driver.AppleUSBHostMergeProperties (1.2) 3845C7DA-337D-35BE-8DA0-E9F83D638E27 <5 3 1>
   22   25 0xffffff8002eda000 0x8c000    0x8c000    com.apple.iokit.IOUSBHostFamily (1.2) 9586526D-BA31-39E5-BC01-F1899C0BD3BE <21 20 19 15 13 8 7 6 5 3 1>
   23    3 0xffffff8003154000 0xf000     0xf000     com.apple.driver.KernelRelayHost (1) 1F7F28E6-FB47-3584-AD39-492ABA531F78 <22 6 5 3>
   24    1 0xffffff80015cf000 0x4f000    0x4f000    com.apple.driver.AppleCredentialManager (1.0) B7AE989D-E5ED-3982-9717-406DB4A98793 <23 18 17 10 8 7 6 5 3 1>
   25    2 0xffffff800163b000 0x6000     0x6000     com.apple.driver.AppleEffaceableStorage (1.0) C5E2FE8D-685F-32A7-A5C9-AFD4F486C1AE <8 6 5 3 1>
   26    0 0xffffff8001645000 0x6000     0x6000     com.apple.driver.AppleFDEKeyStore (28.30) FA91A251-EF61-31DA-8259-ED4C8B562116 <25 10 8 7 6 5 3 1>
   27    2 0xffffff8001b33000 0x7000     0x7000     com.apple.kext.CoreTrust (1) 8DB8E452-ABC9-3042-8840-E19440E8F764 <10 5>
   28    7 0xffffff80017d1000 0x15000    0x15000    com.apple.driver.AppleMobileFileIntegrity (1.0.5) 2D9CF8A5-F82A-3ED8-9ED1-CEDA13D08042 <27 10 8 7 6 5 3 2 1>
   29    5 0xffffff8002bf2000 0x14000    0x14000    com.apple.iokit.IOSCSIArchitectureModelFamily (436) 97C47CDA-654B-39F1-85B0-604D7A4AF221 <7 6 5 3 1>
   30   13 0xffffff8002cc3000 0x12000    0x12000    com.apple.iokit.IOStorageFamily (2.1) 4C895514-8BCD-3820-B320-BF79ED6CBC08 <8 7 6 5 3 1>
   31    2 0xffffff8002c16000 0xe000     0xe000     com.apple.iokit.IOSCSIBlockCommandsDevice (436) 829C1921-497E-3CD0-8562-8898AAC33F7E <30 29 7 6 5 3 1>
   32    2 0xffffff8003130000 0x1a000    0x1a000    com.apple.iokit.IOUSBMassStorageDriver (184) 594CC5FB-A6A8-39E1-9B1A-316BBD4E1310 <30 29 22 14 7 6 5 3 1>
   33    1 0xffffff80018b6000 0x8000     0x8000     com.apple.driver.AppleUSBTDM (511) 0794592D-07A1-3090-AE7A-199B9C25EC69 <32 31 30 29 28 22 8 7 6 5 3 1>
   34    0 0xffffff800172d000 0x7f000    0x7f000    com.apple.driver.AppleKeyStore (2) CFE55699-262D-36F3-AFC2-B0D7D8E3647C <33 28 27 24 23 18 17 10 8 7 6 5 3 1>
   35    0 0xffffff8001880000 0x7000     0x7000     com.apple.driver.AppleSSE (1.0) 8A39875B-278F-3232-8BEE-BFA57F595A83 <23 18 17 8 6 5 3 1>
   36    1 0xffffff8001a5e000 0x7000     0x7000     com.apple.iokit.CoreAnalyticsFamily (1) FD6395D6-F4B2-38CA-AAE5-4E45C7DDB345 <8 7 6 5 3 1>
   37    0 0xffffff8001b3b000 0x4b000    0x4b000    com.apple.iokit.EndpointSecurity (1) E403684A-DB43-3E31-ACF2-6A995DAF6DE0 <28 8 7 6 5 3 2 1>
[... cut ...]
  175    0 0xffffff7fa06fc000 0x8000     0x8000     com.apple.filesystems.autofs (3.0) 6440D13B-A94A-3E7D-95D8-577676241259 <174 8 7 6 5 3 2 1>
  176    0 0xffffff7f9fad4000 0x3000     0x3000     com.apple.driver.AppleUpstreamUserClient (3.6.8) B4DF2CDD-E80F-3E16-9497-19C115CA0DB3 <119 14 13 8 6 5 3 1>
  177    1 0xffffff7fa072f000 0xe000     0xe000     com.apple.filesystems.msdosfs (1.10) 2289D151-AF45-3FE1-B8A7-0DAEC2E05E31 <8 6 5 1>
  181    0 0xffffff7f9fa6b000 0x4b000    0x4b000    com.apple.driver.AppleUSBAudio (400.25.1) 50D8D10F-FA26-35C6-98AA-8F4136ED2808 <105 94 22 21 14 6 5 3 1>
```

chris1111, 2020-08-27:
178, 179, 180,  not there Anyway maybe RtWlanDisk.kext is not loading anymore but this does not prevent it from working with the other two kexts, it proves that I have the same error as you and it works here

I was going to do another test on my macBook without this kext and if it works as well I will simply remove it from the program
Another thing is that Hackintosh with Bootloader injection works without this kext because it has never been included in the program for Hackintosh

CharlieWhile13, 2020-08-27:
Weird, I'll be happy to try any fixes you may have

chris1111, 2020-08-27:
> Weird, I'll be happy to try any fixes you may have

Check verry soon today I probably Update the program

chris1111, 2020-08-27:
V5 is release its work 

[image]

closed 2020-08-27 by chris1111

CharlieWhile13, 2020-08-27:
Still says `No USB-WiFi plug in` in the status bar. Running `kmutil` on both kext files they run fine. They aren't there when using `kextstat`

chris1111, 2020-08-27:
somethings wrong with your Mac man

CharlieWhile13, 2020-08-27:
lol, I know the card works because it worked when I was on my Linux partition.

CharlieWhile13, 2020-08-27:
And it's actually being detected by the Mac, just not by the driver, which is odd

[image]

chris1111, 2020-08-27:
> lol, I know the card works because it worked when I was on my Linux partition.

Work on linux and Mac is different so Linux work good but not your Beta 11
Its work here on 4 macOS 11 beta 5 do you need more picture of all macOS ?

CharlieWhile13, 2020-08-27:
it didn't work on the latest version of Catalina for me either, using the other driver

chris1111, 2020-08-27:
> it didn't work on the latest version of Catalina for me either, using the other driver

What is the adapter brand and name, exact model

CharlieWhile13, 2020-08-27:
Alfa AWUS036ACH, 802.11ac, RTL8812AU

chris1111, 2020-08-27:
> Alfa AWUS036ACH, 802.11ac, RTL8812AU

This is suppose to work

CharlieWhile13, 2020-08-27:
That's what I thought too. The Alfa website links a version of this driver on their site

chris1111, 2020-08-27:
and its not work on 10.15 with the ALPHA driver

CharlieWhile13, 2020-08-27:
Correct

CharlieWhile13, 2020-08-27:
The Alfa driver is your version for versions other than Big Sur, it's basically a mirror

CharlieWhile13, 2020-08-27:
When running kextstat, they still aren't loading for me, running `kextstat | grep -v com.apple` shows nothing

chris1111, 2020-08-27:
> The Alfa driver is your version for versions other than Big Sur, it's basically a mirror

No sir Alpha is V-1830.20.b31.10312018, and mine is V- 1830.32.b17.03122020,

chris1111, 2020-08-27:
> _Basically_, it does seem a little outdated

Alpha drivers is the first Realtek driver for 10.15, they have not updated the driver and utility

chris1111, 2020-08-27:
I have 4 different brand adapter working with the latest release V5

CharlieWhile13, 2020-08-27:
Weird, so you're suggesting it's a problem with my MacBook?

chris1111, 2020-08-27:
> Weird, so you're suggesting it's a problem with my MacBook?

Problem with your Big Sur Beta 5 of course not the macBook

CharlieWhile13, 2020-08-27:
It didn't appear as a USB device on Virtualbox Catalina

chris1111, 2020-08-27:
> It didn't appear as a USB device on Virtualbox Catalina

I don't know much about VMs

CharlieWhile13, 2020-08-27:
Let me try a fresh Catalina install using your driver

chris1111, 2020-08-27:
> Let me try a fresh Catalina install using your driver

https://github.com/chris1111/Wireless-USB-Adapter

CharlieWhile13, 2020-08-27:
Yes sir using that driver, on a complete fresh install of Catalina it worked. 

So my guess is it's just Big Sur being weird in beta with loading the kexts

chris1111, 2020-08-27:
> Yes sir using that driver, on a complete fresh install of Catalina it worked.
> 
> So my guess is it's just Big Sur being weird in beta with loading the kexts

Just try update to BS beta 4 from Cata

CharlieWhile13, 2020-08-27:
Will do probably tomorrow, on a duplicated disk, cause I might just stick on Catalina