chris1111/Wireless-USB-Big-Sur-Adapter #16: Not Working with Big Sur Beta 6 (2015 early MBP)

TamannaWD, 2020-09-14:
Not Working with Big Sur Beta 6 (2015 early MBP). I try all step exactly with your tutorial but no luck, same hardware was perfect on my old Mojave.

Device hardware detect OK without any driver but not pulling on or connect at last. here is my hard ware info on other windows pc 
=======
Manufacturer:	Realtek Semiconductor Corp.
Description:	Realtek 8188GU Wireless LAN 802.11n USB NIC
Driver version:	1040.1.1205.2017
========

my current situation No USB Wifi Plugin in Status Bar on MacBook Pro 11.0 beta 20A5364e
but hard ware show in System Info USB 
02.11n WLAN Adapter:

==========
  Product ID:	0xb711
  Vendor ID:	0x0bda  (Realtek Semiconductor Corp.)
  Version:	2.00
  Serial Number:	00E04CB82101
  Speed:	Up to 480 Mb/s
  Manufacturer:	Realtek
  Location ID:	0x14200000 / 2
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0
==========
[image]
[image]

chris1111, 2020-09-14:
Sorry but 8188GU is not support in macOS only in windows 😔

closed 2020-09-14 by chris1111

TamannaWD, 2020-09-14:
Same hardware was perfect on my old Mojave.

I have another device is it support?
 Manufacturer:	Realtek Semiconductor Corp.
Description:	Realtek RTL8188FTV Wireless LAN 802.11n USB 2.0 Network Adapter #2

chris1111, 2020-09-14:
> Same hardware was perfect on my old Mojave.
> 
> I have another device is it support?
> Manufacturer: Realtek Semiconductor Corp.
> Description: Realtek RTL8188FTV Wireless LAN 802.11n USB 2.0 Network Adapter #2

View support adapter RTL8188FTV is not on the list sorry
## Support Drivers: (Adapter)
<details> 
  <summary>View list  </summary>
	
- ASUS_USB-N10E_92CU

- ASUS_USB-N13_92CU

- ASUS_USB-N10_92CU

- ASUS_1870_8812BU

- ASUS_USB-N10E_92CU

- ASUS_USB-N10_92CU

- ASUS_USB-N13_92CU

- ASUS_USB-AC53_8812BU

- ASUS_USB-AC55B1_8812BU

- ASUS_USB-AC56_8812AU

- ASUS_USB-AC55_8812BU

- ASUS_USB-AC68ALL_8814AU

- ASUS_USB-AC68CE_8814AU

- ASUS_USB-AC68FCC_8814AU

- AboCom_8178_92CU

- AboCom_0811_8811AU

- AboCom_8189_92CU

- AboCom_92EU

- AboCom_88EU

- AboCom_AC_8812AU

- AboCom_AC_8812AU

- Actiontec_8811AU

- AirTies_Air2520_8811AU

- AirTies_Air2525_8811AU

- AboCom_8178_92CU

- AboCom_8189_92CU

- Actiontec_8105_SingleBand_8811AU

- Actiontec_8108_DualBand_8811AU

- Amigo_92CU

- Amigo_92CU

- AzureWave_92CU

- Belkin_1004_92CU

- Belkin_1102_92CU

- Belkin_2102_92CU

- Belkin_2103_92CU

- Belkin_92DUVS_1105

- Belkin_92DUVS_110A

- Belkin_92DUVS_120A

- Belkin_F9L1106_v2_8812AU

- Belkin_F9L1106v2_8812AU

- Buffallo_25D_8812AU

- Buffallo_433DM_8811AU

- Buffallo_WI_U2_433DHP_8811AU

- Buffallo_WLP_U2_433DHP_8811AU

- Compare-8010_92CU

- Compare-8011_92CU

- Corega_92CU

- DLink_DWA121_92CU

- DLink_DWA123_92CU

- DLink_DWA131B1_92CU

- DLink_DWA132_92CU

- DLink_DWA133_92CU

- DLink_DWA123_88EU

- DLink_DWA125_88EU

- DLink_DWA131C1_92EU

- DLink_DWA131E_92EU

- DLink_DWA171_8812AU

- DLink_DWA182B1_8812AU

- DLink_DWA182_8812AU

- DLink_DWA192_8814AU

- DLink_GO_USB_N150_88EU

- ELECOM_WDC300SU2S_92CU

- ELECOM_8811AU

- ELECOM_WDB433SU2M_8811AU

- ELECOM_WDC1300DU3_8814AU

- ELECOM_WDC1300SU3_8814AU

- ELECOM_WDC150SU2M_88EU

- ELECOM_WDC433DU2_8812AU

- ELECOM_WDC433SU2M2_8811AU

- EDIMAX- EW-7722UTn V2 

- EDIMAX N300

- EDIMAX EW-7811Un

- Edimax_AC1750_8814AU

- Edimax_AC1750_A834_8814AU

- Edimax_AC600_8812AU

- Edimax_EW-7611ULB_8723BU

- Edimax_EW-7811UAC_8812AU

- Edimax_EW-7822UAC_8812AU

- Edimax_EW-7822ULC_8812AU

- Edimax_GLP_8812AU

- Edimax_7811_92CU

- Edimax_7822_92CU

- Feixun_90_92CU

- Feixun_91_92CU

- EnGenius_AC_8812AU

- HP_92CU

- Hawking_HWDN3_92CU

- Hawking_HWUN4_92CU

- Hercules_HWUm300_92CU

- Hercules_HWUp150_92CU

- Hawking_8812AU

- Hawking_HW7ACU_8812AU

- IO_DATA_AC433UM_8812AU

- O_DATA_WN-AC867U_8812AU

- Infocus_INA-LCKEY_8812AU

- IO_DATA_92CU

- Linksys_WUSB6300_8812AU

- Logitec_92CU

- Loopcomm_ACA1_8812AU

- Netgear_A7000

- Netgear_N300MA_92CU

- Netgear_WNA1000M_92CU

- Netgear_WNA3100M_92CU

- Netgear_A6100_8812AU

- Netgear_A6200v2_8812AU

- PCI_BT-Micro3H2X_92CU

- PCI_GW_USEco300_92CU

- PCI_GW_USLight_92CU

- PCI_GW_USNano2_92CU

- PCI_GW_USValue_EZ_92CU

- PCI_SW_WF02-AD15_92CU

- PCI_GW-300S_92EU

- PCI_GW-450S_8812AU

- PCI_GW-900D_8812AU

- Proxim_USB-9100_8812AU

- RTL8188CTV

- RTL8188CTV_0A8A

- RTL8188CTV_8011

- RTL8188CU

- RTL8188CUS_1E1E

- RTL8188CUS_2E2E

- RTL8188CUS_5088

- RTL8188CUS_Combo

- RTL8188CUS_Combo_AFF8

- RTL8188CUS_Combo_AFFB

- RTL8188CUS_Combo_AFFC

- RTL8188CUS_Solo

- RTL8188CUS_VL

- RTL8188CUS_solo_AFF7

- RTL8188CUS_solo_AFF9

- RTL8188CUS_solo_AFFA

- RTL8188RU

- RTL8188RU_Netcore

- RTL8192CU

- RTL8192CU_8177

- RTL8192CU_8178

- RTL8192DU_VS

- RTL8188EU

- RTL8188EU_ETV

- RTL8188EU_VAU

- RTL8192EU

- RTL8192EU-2

- RTL8811AU

- RTL8812AU

- RTL8812AU-VL

- RTL8812AU-VN

- RTL8812AU-VS

- RTL8814AU

- Sitecom_WL365_92CU

- Sitecom_WLA1001v1_92CU

- Sitecom_WLA2102_92CU

- Sitecom_WLA4001_92CU

- Sitecom_WLA1100_88EU

- Sitecom_WLA2104_8812AU

- Sitecom_WLA7100_8812AU

- Sitecom_WLA8100_8814AU

- TPLink-Archer_T2U_NANO

- TL-WN823Nv3

- TL-WN725Nv3

- TL-WN723Nv3

- TL-WN722Nv3

- TL-WN821Nv6

- TPLink_92CU

- TPLink_821v5_92EU

- TPLink_822v4_92EU

- TPLink_823v2_92EU

- TPLink_8812AU_1

- TPLink_8812AU_2

- TPLink_8812AU_3

- TPLink_88EUSU

- TPLink_T4UH_8812AU

- TPLink_T4U_8812AU

- TPLink_T9UH_8814AU

- TRENDnet N150 Micro

- Trendnet_624D_92CU

- Trendnet_648B_92CU

- Trendnet_92DUVS

- TrendNet_TEW804B_8812AU

- TrendNet_TEW805B_8812AU

- TrendNet_TEW809UB_8814AU

- Western_AC_8812AU

- ZyXEL_AC_8812AU

- ZyXEL_92CU
				

</details>

TamannaWD, 2020-09-15:
I use this device perfectly on my MacBook Pro Mojave long time no issue. But last 2 day ago update it to Big Sur device Unsupported

Realtek 8188GU Wireless LAN 802.11n USB NIC

Can you please make an update with this device support or help me , my builtin WiFi hardware damage so i realy need it urgent dear.

TamannaWD, 2020-09-17:
> Sorry but 8188GU is not su

Lol, its Working fine in my Mac with TP LINK WN722N Software
[image]