chris1111/Wireless-USB-Big-Sur-Adapter #17: cannot boot after installing Wireless-USB-OC-Big-Sur-Adapter

skandavc18, 2020-10-01:
I have WN725N V3 adapter. Used to work in my Catalina Hackintosh. But after upgrading to Big Sur beta, it is not working. 

When I use Wireless-USB-OC-Big-Sur-Adapter and add the 2 kexts to config.plist as in the video and restart, it does not boot at all.

When I use Wireless-USB-Big-Sur-Adapter, it says "NO USB Wifi Device Plugged in". Tried disabling SIP and re-enabling and then disabling again and installing.  But not working. 

Please help!

I have attached the log file for Wireless-USB-Big-Sur-Adapter pkg
[Installer Log 1-Oct-2020.txt](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/5312516/Installer.Log.1-Oct-2020.txt)

skandavc18, 2020-10-01:
No. what I mean is that I first experimented with Wireless-USB-OC-Big-Sur-Adapter program. It didn't work. So I uninstalled it. Later I tried Wireless-USB-Big-Sur-Adapter program. Even that didn't work. I didn't mix both.

chris1111, 2020-10-01:
Wireless-USB-OC-Big-Sur-Adapter is support to easily on your Hackintosh, if not check your OpenCore Setup

skandavc18, 2020-10-01:
ok.

skandavc18, 2020-10-01:
Got it Working!
As you rightly said, the problem was with my OC configuration. I had left the secureBootModel option in Misc/Security in OC as default. It has to be set as disabled. This made it work. 

Thanks for the driver. It's fantastic!

chris1111, 2020-10-01:
> Got it Working!
> As you rightly said, the problem was with my OC configuration. I had left the secureBootModel option in Misc/Security in OC as default. It has to be set as disabled. This made it work.
> 
> Thanks for the driver. It's fantastic!
> 
> Also, your driver works perfectly with even TPLink WN725N V2 as well

Glad to hear it but if you came back again with a failure that's what I was going to tell you because it could only be that
Also tt is better to use the kexts injection by OpenCore 😁 Thanks enjoy