chris1111/Wireless-USB-Big-Sur-Adapter #22: Archer T2U Nano not recognized

nfriacowboy, 2020-11-13:
[Registo do Instalador 13-Nov-2020.txt](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/5536903/Registo.do.Instalador.13-Nov-2020.txt)
Hi,

After install Wireless.USB.Big.Sur.Adapter-V-5 and following the steps on the video I get a message on the StatusBar App that the hardware is not recognized.

I noticed that during installation the "System Extension Update" window never come up and after installation in the "Security & Privacy" in update message.

attached is the installation log, thanks in advance

chris1111, 2020-11-13:
If the System Extension Update" window never come up the SIP is not totaly disable

closed 2020-11-13 by chris1111

nfriacowboy, 2020-11-13:
Correct!

I followed the instructions here https://www.imore.com/how-turn-system-integrity-protection-macos
and reinstall the driver again and worked perfectly!!! Thanks a lot!!!!

chris1111, 2020-11-13:
> Correct!
> 
> I followed the instructions here https://www.imore.com/how-turn-system-integrity-protection-macos
> and reinstall the driver again and worked perfectly!!! Thanks a lot!!!!

Great 👏