chris1111/Wireless-USB-Big-Sur-Adapter #24: Edimax EW-7822ULC not detected

jeremybeny, 2020-11-16:
hi on my Mac with Big Sur my usb dongle is not detected one the network preferences. I have the little icon with the dongle and a Red Cross on it. is there a way to make it work? thx

jeremybeny, 2020-11-16:
I tried the #6 and #10 post answers by enabling disabling sip and gatekeeper but still not working

chris1111, 2020-11-16:
EW-7822ULC its a known working adapter, it is already test and is support

closed 2020-11-16 by chris1111

jeremybeny, 2020-11-16:
then maybe something went wrong during the install I uninstalled it restarted one recovery , enabled the sip restarded disable the sip installed it and I don't have the realtek option in the system preference, can you help me find were is the problem ? thank you

chris1111, 2020-11-16:
Reset NVRAM
Disable SIP

jeremybeny, 2020-11-16:
thnx will try