chris1111/Wireless-USB-Big-Sur-Adapter #30: making source code available

alikazemkhanloo, 2020-11-27:
Hi,
Thank you for this great project.
Can you provide the source code?
I mean, putting the entire Xcode project on Github (obviously except for credentials, etc), so that anybody can edit and build their own version. This way other developers my help adding features or fixing bugs.

chris1111, 2020-11-27:
I don't have the source code, the only source available is from linux

closed 2020-11-27 by chris1111

stkrzysiak, 2021-09-02:
Hi @chris1111 Thank you for this repo! I have been thinking about the source code aspect of this utility./adapter I think this topic should be revisited. If there is no source code then it'd be ideal to document how this utility was created(linux drivers packaged as kernel extensions using this process, step 1.....). The concern here is that if we disable SIP then we must really trust the source code, which currently we can not see. If we can reproduce the packaging process ourselves then I think it'd be safer for some of us to use.

In the meantime, while researching the topic of the big sur wifi drivers I came across this [thread](https://www.insanelymac.com/forum/topic/299948-usb-wifi-updated-ralinkmediatek-rt2870-rt2770-rt3x7x-rt537x-rt5572/page/8/), which seems to document some, if not all of the work and intent of this repo. Seeing that it was a community effort makes me a bit easier about disabling SIP and just going for it. Thank you @chris1111 and others for putting this together. If it's evident after reading the entire thread how this utility came about maybe I can submit a PR to this repo providing source/transparency.

chris1111, 2021-09-02:
The driver comes from [Realtek](https://www.realtek.com/en/) But the compagnie stop the developement
Its Trust source, no modification its original drivers and Utility
[image]
A lots of wifi compagnie link the User here for using this utility because all of the fabricant drop support on are WIFI adapter