chris1111/Wireless-USB-Big-Sur-Adapter #32: No usb wifi plugin

alfathdirk, 2020-12-03:
No usb wifi plugin after installing and restart
[image]

os: BIG SUR 10.0.1 (real mac not hackintosh)
usb wifi: mt7601 

[image]
usb dongle detected in system profiler
```

802.11 n WLAN:
          Product ID: 0x7601
          Vendor ID: 0x148f
          Version: 0.00
          Serial Number: 1.0
          Speed: Up to 480 Mb/s
          Manufacturer: MediaTek
          Location ID: 0x14100000 / 30
          Current Available (mA): 500
          Current Required (mA): 160
          Extra Operating Current (mA): 0

```

chris1111, 2020-12-03:
MediaTek is not support here 
Its Realtek Drivers

closed 2020-12-03 by chris1111

alfathdirk, 2020-12-03:
Can i use this on os Big Sur ?

https://github.com/chris1111/Wireless-Ralink-Panel-Utility