chris1111/Wireless-USB-Big-Sur-Adapter #36: No USB-WiFi Plugin on Big Sur v11.1

UJPrasad, 2020-12-17:
[image]

[image]

[image]
[image]

chris1111, 2020-12-17:
Adapter brand?

UJPrasad, 2020-12-17:
Hi, @chris1111 thanks for the immediate response.
Adapter brand -> TP-Link Archer T2U Nano
Big Sur Version -> 11.1 (20C69)

chris1111, 2020-12-17:
> Hi, @chris1111 thanks for the immediate response.
> Adapter brand -> TP-Link Archer T2U Nano
> Big Sur Version -> 11.1 (20C69)

Archer T2U Nano is support and already test in BS
make sure the SIP is fully disable if you don't have A popup security for allowing extensions this is because SIP is not fully disable

UJPrasad, 2020-12-18:
I have done this to fully disable SIP:

```
Restart your computer in Recovery mode.

Launch Terminal from the Utilities menu.

Run the command csrutil disable.

Restart your computer.
```

Do we need to do something else or this should work? Is there any other way rather than typing `csrutil status` to check if SIP is fully disabled?

UJPrasad, 2020-12-18:
It's a real mac - MacBook Pro Model A1502.

chris1111, 2020-12-18:
> It's a real mac - MacBook Pro Model A1502.

all my test and video is from the same MacBook Pro so you have a corruption problem or whatever?
try to uninstall the program reboot to the recovery HD enable the SIP reboot to your mac then reboot again in recovery HD disable the SIP and install the program

closed 2020-12-18 by chris1111

Shekara727, 2026-10-06:
hyh