chris1111/Wireless-USB-Big-Sur-Adapter #41: Mac cannot find location

evasilis, 2020-12-25:
Hi there

After installing the V5 and later the V7 drivers, big sur says that it cannot find location as there is no wifi. Would you have any suggestions?

Thanks for your time and work!
V

chris1111, 2020-12-25:
> Hi there
> 
> After installing the V5 and later the V7 drivers, big sur says that it cannot find location as there is no wifi. Would you have any suggestions?
> 
> Thanks for your time and work!
> V

What is the adapter brand?

evasilis, 2020-12-25:
Thanks for your reply! It is a tp-link Archer TU2 Plus connected via USB 2 to a 5,1 mac pro. =Big sur was installed the bigmac 1.1 method and it is unpatched.

Thanks a lot!

chris1111, 2020-12-25:
It is really strange because Archer TU2 Plus is already test here  [Known Compability Adapter](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/issues/18)

chris1111, 2020-12-25:
Do you see this before reboot?

[image]

[image]

evasilis, 2020-12-25:
Yes and I allowed and then rebooted.

chris1111, 2020-12-25:
> Yes and I allowed and then rebooted.

Try another USB port ?
Or
Try Uninstall the reboot then reinstall the program
 Let me known

evasilis, 2020-12-25:
I will do that.

Thanks!

evasilis, 2020-12-25:
I did uninstall - reboot - reinstall and also tried a different usb port but there was no change.

evasilis, 2021-01-01:
Happy new year to you and yours!

chris1111, 2021-01-01:
I wish you Happy New Year

[image]

closed 2021-01-01 by chris1111