chris1111/Wireless-USB-Big-Sur-Adapter #48: After m1 macmini 11.1 disables sip, the installation can be successful, but the network card cannot be recognized

baipiaoing, 2020-12-30:
After m1 macmini 11.1 disables sip, the installation can be successful, but the network card cannot be recognized when the network card is inserted, and "No USB-WiFi Plug in" is still displayed. The network card is confast cf-811ac

chris1111, 2020-12-30:
Why using Cheap USB adapter on M1 macMini with native WIFI

baipiaoing, 2020-12-30:
At my school in China, the campus network requires authentication to log in. I want to use my computer to connect to wifi and open a hot spot

chris1111, 2020-12-30:
I don't think Hot spot work with WIFI adapter but see this
Runing Intel app on Mac M1 chip
https://derflounder.wordpress.com/2020/11/17/installing-rosetta-2-on-apple-silicon-macs/

baipiaoing, 2020-12-30:
I mean, I use this adapter to connect to Wi-Fi and use the built-in network card of the computer to open the hotspot.

baipiaoing, 2020-12-30:
The previous computer did this in 10.15. Now after changing the computer, the first method can no longer be used.

chris1111, 2020-12-30:
No way to do this on M1 chip

closed 2020-12-30 by chris1111

blairmacintyre, 2021-03-18:
is this likely to ever work on an M1 mac?

chris1111, 2021-03-18:
> is this likely to ever work on an M1 mac?

not for now maybe in the futur when Realtek updated the Utility

Tarmstrong95, 2022-12-06:
@chris1111 any update on this?