chris1111/Wireless-USB-Big-Sur-Adapter #49: installation on vmware image of Bug Sur

frguerin, 2020-12-30:
Hello,
i am using a vmware image of big sur and I have a netgear dongle A6100 that I would have liked to connect and use directly from the vm.
Installing the pkg I have an error. Enclosed the installation log.
Not sure this setup (big sur vm image running on vmware workstation on host win10) is supported though...
Regards, François.
[Installer Log 30-Dec-2020.txt](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/5754949/Installer.Log.30-Dec-2020.txt)

chris1111, 2020-12-30:
I looked at your Log, there is stuff that happens in a VM that I can't understand because I know absolutely nothing about VM's, sorry but I can't do anything about it.

frguerin, 2020-12-30:
thanks a lot for looking. Do not bother, I reverted to my snapshot of catalina 10.15 because Big Sur 11 is not working well under vmware with very slow overall performance. I will let a few version go by before attempting again to upgrade. I will now try your kext installation using catalina on my vm and let you know if it works better. Regards, François.

closed 2020-12-30 by frguerin