chris1111/Wireless-USB-Big-Sur-Adapter #50: unable to install the driver on Big Sur version 11.1

lukapril, 2021-01-06:
I've downloaded the adapter-V-7 pkg, trying to install that onto my MacBook Pro running Big Sur version 11.1, the installation failed with message The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance. 

I have a Linksys WUSB6400M adapter.

chris1111, 2021-01-06:
If is called the SIP is not disable

closed 2021-01-06 by chris1111

lukapril, 2021-01-06:
Thanks for the quick response. It was my bad, I disabled SIP and able to install successfully. Does SIP has to stay disabled in order for the driver to work? (ie can I enable SIP after the installation?)

chris1111, 2021-01-06:
> @chris1111 Thanks for the quick response. It was my bad, I disabled SIP and able to install successfully. Does SIP has to stay disabled in order for the driver to work? (ie can I enable SIP after the installation?)

No SIP must be disable since Beta 5