chris1111/Wireless-USB-Big-Sur-Adapter #54: Installation package fails on Big Sur (Mac OS X 11.1)

RickD123, 2021-01-20:
Hi,

I bought a TP-Link Archer T3U/Plus to be installed on my Mac running Big Sur (Mac OS X 11). Even after disabling Gatekeeper and following the steps outlined in the video to install the software, the installation routine just fails at the end of the installation process with "Installation failed". Any idea on may be causing this and how to troubleshoot it?

Many thanks.

Kind regards
Sebastian

RickD123, 2021-01-20:
Can closed closed. SIP was still enabled, disabling it allowed me to install the software.

chris1111, 2021-01-20:
And its work for your TP-Link Archer T3U/Plus

RickD123, 2021-01-20:
The installation succeeded, but the USB Wifi device is not connected error occurs. I can see the device in the system information but I cannot get it working.

RickD123, 2021-01-20:
¶
[image]

RickD123, 2021-01-20:
[image]

Now with a screenshot

RickD123, 2021-01-20:
In system information, the device is listed under USB, 802.11ac NIC (sorry in German):

Produkt-ID:	0x0138
  Hersteller-ID:	0x2357
  Version:	2.10
  Seriennummer:	123456
  Geschwindigkeit:	Bis zu 480 MBit/s
  Hersteller:	Realtek
  Standort-ID:	0x14300000 / 1
  Verfügbare Stromstärke (mA):	500
  Erforderliche Stromstärke (mA):	500
  Zusätzlicher Betriebsstrom (mA):	0

RickD123, 2021-01-20:
Any ideas?

chris1111, 2021-01-20:
> Any ideas?

Try 
sudo kmutil install --update-all
sudo kcditto

then reboot

RickD123, 2021-01-20:
Result is the same. How do I uninstall? I will then reboot, re-enable SIP, reboot, uninstall & reboot again, disable SIP, reboot, install again. Maybe this works.

chris1111, 2021-01-20:
Yes uninstall

RickD123, 2021-01-20:
Sorry for asking a stupid question but how do I uninstall? The app is not listed in my "Applications" folder

chris1111, 2021-01-20:
try install you will see

chris1111, 2021-01-20:
What is your Mac ?
is it a Hackintosh or a real Mac ?

RickD123, 2021-01-20:
Mac, 2018 Mac Mini

chris1111, 2021-01-20:
> Mac, 2018 Mac Mini

no wifi Apple native?

RickD123, 2021-01-20:
I have Unifi APs and while they work fine with most of the iPhones in the house, they do not work well with my Mac mini. I have   random disconnects, slow connections etc., which are known issues also described in the Unifi forums. I guess they are related to either the chipset of the Mac Mini or the lack of WEP3 support in the Unifi APs. Now I am trying to figure out alternative/better way of doing Wifi with my Mac.

chriskitz24, 2021-02-04:
Thanks A LOT! I hope and assume that re-enabling SIP after getting this to work shouldn't be an issue? I'm going to try that right now! but as of this comment both my adapters are working, THANKS AGAIN!!!

chriskitz24, 2021-02-04:
**NOTE** I did turn SIP back **ON** and the adapters **DON'T** work, so keep that in mind.... thanks for the solution though since TP-Link can't figure their sh*t out ❤️

AdnanKhan786, 2023-07-23:
Guys can please anyone tell me how can I get tp link T3U plus wifi adapter drivers for my macos monterey 12.6.3