chris1111/Wireless-USB-Big-Sur-Adapter #57: Issue with using Adapter

chriskitz24, 2021-01-23:
I have a MacBook Pro 13in.- 2017 with 4 thunderbolt (or USB-C) ports. I have to use a USB-C to USB-A adapter from apple to use any USB device... and I can see it under System Information>USB as being recognized, but so far I cannot get the adapter to work. wondering if this is because of the USB-C to USB-A adapter.

I have these two adapters:

-Archer T2U Plus
-Archer T3U 
BOTH by TP-Link

Please get back to me kitzmanc@gmail.com

chriskitz24, 2021-01-23:
> I have a MacBook Pro 13in.- 2017 with 4 thunderbolt (or USB-C) ports. I have to use a USB-C to USB-A adapter from apple to use any USB device... and I can see it under System Information>USB as being recognized, but so far I cannot get the adapter to work. wondering if this is because of the USB-C to USB-A adapter.
> 
> I have these two adapters:
> 
> -Archer T2U Plus
> -Archer T3U
> BOTH by TP-Link
> 
> Please get back to me [kitzmanc@gmail.com](mailto:kitzmanc@gmail.com)

I've also tried the whole run-around about installing and removing the original driver. and the issue is that MacOS 11.1 has been a nightmare for updates, please help us

RickD123, 2021-01-23:
See my own case for details on how I solved it for the Archer T2U Plus: https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/issues/54

closed 2021-01-28 by chris1111