chris1111/Wireless-USB-Big-Sur-Adapter #58: Wrong password, minutes after proper connection with same password

ts1g, 2021-01-25:
This issue has cropped up for me when using both the Mac native (v7) and OC-version (v6) of the BigSur Wireless adapter release:

1. Start up machine, wait for wireless adapter icon to show up in menu bar.
2. Click on wireless adapter icon,  choose an AP and put in password for AP
3. Work online using the wireless adapter connected to the AP.
4. A dialogue box appears “Wrong password. Please try again.” (Some times it happens a few minutes after I’ve connected and have been using the wireless adapter / AP combination)
5. Click dialogue box, go back to wireless adapter icon, choose AP (don’t need to put in password this time) and continue working with same wireless adapter / AP combination

If it only happened once per hour, no big deal, but since it happens once every few minutes, I tried fully uninstalling, made sure SIP is off in OC version, dropped back to Big Sur 11.0.1 instead of 11.1, used a different wireless adapter. Yet the random disconnect / wrong password persists.

chris1111, 2021-01-25:
nerver see this

closed 2021-01-25 by chris1111

ts1g, 2021-01-25:
@chris1111 Just because you’ve never seen it, are you saying it’s not an issue? I can replicate it on multiple machines and with multiple adapters