chris1111/Wireless-USB-Big-Sur-Adapter #71: BT Wi-Fi Authentication Page is not showing

jakemike1, 2021-02-26:
Hi,

First of all I would like to say thank you for doing great job on making Wi-Fi adapters to work on Big Sur.

I am having issue when connecting to BT Wi-Fi hotspot. When connected I should see the authentication webpage. The problem is that the page is not showing when connected using the TP-Link. I tried connecting to authentication page (captive.apple.com and www.btwifi.co.uk:8443) manually but the pages are not loading when using the TP-Link adapter.

Any ideas please?

chris1111, 2021-02-26:
The page not loading using network cable 😂
[image]

jakemike1, 2021-02-26:
Are you connected to the BT Wi-fi Hotspot?

chris1111, 2021-02-26:
no but BT USB Realtek adapter not working in macOS , BT never working

closed 2021-02-26 by chris1111

jakemike1, 2021-02-26:
BT is a company name 😉 not Bluetooth. Sorry, I should have mentioned that in the first place.

It’s like connecting to public unsecured wifi hotspot, when you connect the window pops up with authentication page where you type in you credentials. Just like at the airport, hotel or any other public WiFi’s.

The problem is that the authentication page does not appear when using the Archer with your drivers and I am unable to connect to the page manually.

chris1111, 2021-02-26:
I don't think that's working you need to address this issue at Realtek or TP Link because even with the TP Link drivers this is not working I am pretty sure somme people already post issue about that