chris1111/Wireless-USB-Big-Sur-Adapter #72: wifi dongle adapter (rt5370) not working in BS11.1. Worked in catalina

neuberfran, 2021-02-26:
Hi,

https://docs.google.com/document/d/1zRFudK9icFa2KaejitafkHqLBSYzcqrCCckCBC2U6SY/edit

I have ralink rt5370 (from lsusb in ubuntu) here. In hackintosh catalina it worked.

In BS11.1 it not work. Why? 

1) i disable the SecureBootModel 
2) I disable the Sip

[image]

chris1111, 2021-02-26:
ralink rt5370 Its a Mediatek drivers, this is not working after High Sierra 10.13
Not support here 🤭

closed 2021-02-26 by chris1111

neuberfran, 2021-02-26:
@chris1111 As I said above, rt5370 was running normally at 10.15.7.

chris1111, 2021-02-26:
> @chris1111 As I said above, rt5370 was running normally at 10.15.7.

https://www.mediatek.com/products/broadbandwifi/rt5370

neuberfran, 2021-02-28:
@chris1111  pls. My last question is: From the list of compatible wifi usb adapters on your github (with BS11.1, of course) which is the cheapest?