chris1111/Wireless-USB-Big-Sur-Adapter #73: Error with Archer T2U Plus and MacOS 11.2

quanhinh9, 2021-02-28:
I use MacOS 11.2 and Archer T2U Plus. I install Wireless USB Big Sur Adapter-V7.pkg follow link https://community.tp-link.com/en/home/stories/detail/1642, but it not working.

Does the Wireless USB Big Sur Adapter-V7.pkg not support MacOS 11.2?

chris1111, 2021-02-28:
The program support BS 11.2 .2 and the T2U Plus work is already testing
[image]
[image]

closed 2021-02-28 by chris1111

quanhinh9, 2021-02-28:
After install Wireless USB Big Sur Adapter-V7.pkg, click WIFI icon: no USB-Wifi Plug in.

I run sudo kmutil install --update-all

checking collections...
updated extension 'com.apple.driver.usb.AppleUSBVHCICommonRSM' found on disk at '/System/Library/Extensions/AppleUSBCommon.kext/Contents/PlugIns/AppleUSBVHCICommonRSM.kext' (mach-o UUID changed from <unknown> to 0BF3D860-086D-3FB9-9EA6-A98CB08D881B)
updated extension 'com.apple.nke.rvi' found on disk at '/Library/Apple/System/Library/Extensions/RemoteVirtualInterface.kext' (mach-o UUID changed from <unknown> to 5608F5DD-5A33-375D-B4CE-2F28C91DEDF4)
rebuilding release collections: boot, system
rebuilding release collections:
	boot kernel collection
	system kext collection
failed to build release collection:
	Error Domain=NSPOSIXErrorDomain Code=30 "Read-only file system"
Error Domain=KMErrorDomain Code=71 "Internal error: The operation couldn’t be completed. Read-only file system" UserInfo={NSLocalizedDescription=Internal error: The operation couldn’t be completed. Read-only file system}

How to fix it?