chris1111/Wireless-USB-Big-Sur-Adapter #85: after update stopped working

RazaGR, 2021-04-24:
I just did an upgrade big sur to 11.4 and was not able to see that adapter is attached, I uninstalled and reinstalled by disabling SIP, but no success.

chris1111, 2021-04-24:
adapter?

closed 2021-04-24 by chris1111

RazaGR, 2021-04-24:
adapter, I meant Usb device, I tried other ports but no success. I could see in System Report that USB device is listed:

```
802.11ac NIC:
  Product ID:	0x012d
  Vendor ID:	0x2357
  Version:	2.10
  Serial Number:	123456
  Speed:	Up to 480 Mb/s
  Manufacturer:	Realtek
  Location ID:	0x14410000 / 28
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0
```
but no lights are being turned on , and under icon error message is "No USB-WiFi plug in"
its TP-Link WiFi USB Adapter AC1300 Archer T3U  dongle.

RazaGR, 2021-04-24:
installer log
```
Apr 24 07:18:22 RazaMac Installer[1417]: @(#)PROGRAM:Install  PROJECT:Install-1000
Apr 24 07:18:22 RazaMac Installer[1417]: @(#)PROGRAM:Installer  PROJECT:Installer-1020
Apr 24 07:18:22 RazaMac Installer[1417]: Hardware: MacBookPro14,3 @ 2.80 GHz (x 8), 16384 MB RAM
Apr 24 07:18:22 RazaMac Installer[1417]: Running OS Build: macOS 11.4 (20F5046g)
Apr 24 07:18:22 RazaMac Installer[1417]: Env: USER=raza
Apr 24 07:18:22 RazaMac Installer[1417]: Env: __CFBundleIdentifier=com.apple.installer
Apr 24 07:18:22 RazaMac Installer[1417]: Env: COMMAND_MODE=unix2003
Apr 24 07:18:22 RazaMac Installer[1417]: Env: LOGNAME=raza
Apr 24 07:18:22 RazaMac Installer[1417]: Env: PATH=/usr/bin:/bin:/usr/sbin:/sbin
Apr 24 07:18:22 RazaMac Installer[1417]: Env: APP_SANDBOX_CONTAINER_ID=com.apple.Safari
Apr 24 07:18:22 RazaMac Installer[1417]: Env: SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.ewtYTptwm4/Listeners
Apr 24 07:18:22 RazaMac Installer[1417]: Env: SHELL=/bin/zsh
Apr 24 07:18:22 RazaMac Installer[1417]: Env: __CF_USER_TEXT_ENCODING=0x1F5:0x0:0x0
Apr 24 07:18:22 RazaMac Installer[1417]: Env: XPC_SERVICE_NAME=application.com.apple.installer.1152921500312163230.1152921500312163235
Apr 24 07:18:22 RazaMac Installer[1417]: Env: XPC_FLAGS=0x0
Apr 24 07:18:22 RazaMac Installer[1417]: Env: HOME=/Users/raza
Apr 24 07:18:22 RazaMac Installer[1417]: Env: TMPDIR=/var/folders/7r/b64w_9bs10dcgr0lsn9tg5qh0000gn/T/
Apr 24 07:18:22 RazaMac Installer[1417]: Wireless USB Big Sur Adapter-V-7  Installation Log
Apr 24 07:18:22 RazaMac Installer[1417]: Opened from: /Users/raza/Downloads/Wireless USB Big Sur Adapter-V-7.pkg
Apr 24 07:18:22 RazaMac Installer[1417]: Package Authoring Error: <background> has an unsupported MIME type: image/data
Apr 24 07:18:22 RazaMac Installer[1417]: Package Authoring Error: <background-darkAqua> has an unsupported MIME type: image/data
Apr 24 07:18:22 RazaMac Installer[1417]: Package Authoring Error: <background_scaling> has an unsupported MIME type: X-NSObject/NSNumber
Apr 24 07:18:22 RazaMac Installer[1417]: Package Authoring Error: <background_alignment> has an unsupported MIME type: X-NSObject/NSNumber
Apr 24 07:18:22 RazaMac Installer[1417]: Package Authoring Error: <layout-direction> has an unsupported MIME type: X-NSObject/NSNumber
Apr 24 07:18:22 RazaMac Installer[1417]: Product archive /Users/raza/Downloads/Wireless USB Big Sur Adapter-V-7.pkg trustLevel=100
Apr 24 07:18:22 RazaMac Installer[1417]: External component packages (5) trustLevel=100 (trust evaluation failed: Error Domain=PKInstallErrorDomain Code=101 "The package “Wireless USB Big Sur Adapter-V-7” is not signed." UserInfo={NSURL=#uninstall.pkg -- file:///Users/raza/Downloads/Wireless%20USB%20Big%20Sur%20Adapter-V-7.pkg#Distribution, PKInstallPackageIdentifier=com.109Driver.wirelessUsbBigSurAdapterv7.uninstall.pkg, NSLocalizedDescription=The package “Wireless USB Big Sur Adapter-V-7” is not signed.})
Apr 24 07:18:22 RazaMac Installer[1417]: Could not load resource license: (null)
Apr 24 07:18:23 RazaMac Installer[1417]: Package Authoring Error: Infinite loop between <choice> start_* attributes and mixed state aborted.
Apr 24 07:18:23 RazaMac Installer[1417]: Package Authoring Error: Infinite loop between <choice> attributes and mixed state aborted.
Apr 24 07:18:31 RazaMac Installer[1417]: ================================================================================
Apr 24 07:18:31 RazaMac Installer[1417]: User picked Custom Install
Apr 24 07:18:31 RazaMac Installer[1417]: Choices selected for installation:
Apr 24 07:18:31 RazaMac Installer[1417]: 	Upgrade: "Wireless USB Big Sur Adapter-V-7"
Apr 24 07:18:31 RazaMac Installer[1417]: 	Upgrade: "Wireless USB Big Sur Adapter"
Apr 24 07:18:31 RazaMac Installer[1417]: 		Wireless USB Big Sur Adapter-V-7.pkg#statusbarapp.pkg : com.109Driver.wirelessUsbBigSurAdapterv7.statusbarapp.pkg : 1.0
Apr 24 07:18:31 RazaMac Installer[1417]: 		Wireless USB Big Sur Adapter-V-7.pkg#agents.pkg : com.109Driver.wirelessUsbBigSurAdapterv7.Agents.pkg : 1.0
Apr 24 07:18:31 RazaMac Installer[1417]: 		Wireless USB Big Sur Adapter-V-7.pkg#daemon.pkg : com.109Driver.wirelessUsbBigSurAdapterv7.Daemon.pkg : 1.0
Apr 24 07:18:31 RazaMac Installer[1417]: 		Wireless USB Big Sur Adapter-V-7.pkg#drivers.pkg : com.109Driver.wirelessUsbBigSurAdapterv7.Drivers.pkg : 1.0
Apr 24 07:18:31 RazaMac Installer[1417]: ================================================================================
Apr 24 07:18:31 RazaMac Installer[1417]: It took 0.00 seconds to summarize the package selections.
Apr 24 07:18:36 RazaMac Installer[1417]: Administrator authorization granted.
Apr 24 07:18:36 RazaMac Installer[1417]: ================================================================================
Apr 24 07:18:36 RazaMac Installer[1417]: User picked Standard Install
Apr 24 07:18:36 RazaMac Installer[1417]: Choices selected for installation:
Apr 24 07:18:36 RazaMac Installer[1417]: 	Upgrade: "Wireless USB Big Sur Adapter-V-7"
Apr 24 07:18:36 RazaMac Installer[1417]: 	Upgrade: "Wireless USB Big Sur Adapter"
Apr 24 07:18:36 RazaMac Installer[1417]: 		Wireless USB Big Sur Adapter-V-7.pkg#statusbarapp.pkg : com.109Driver.wirelessUsbBigSurAdapterv7.statusbarapp.pkg : 1.0
Apr 24 07:18:36 RazaMac Installer[1417]: 		Wireless USB Big Sur Adapter-V-7.pkg#agents.pkg : com.109Driver.wirelessUsbBigSurAdapterv7.Agents.pkg : 1.0
[... cut ...]
Apr 24 07:19:03 RazaMac Installer[1417]:        disk      0.03 seconds
Apr 24 07:19:03 RazaMac Installer[1417]:      script      0.00 seconds
Apr 24 07:19:03 RazaMac Installer[1417]:        zero      0.00 seconds
Apr 24 07:19:03 RazaMac Installer[1417]:     install      27.15 seconds
Apr 24 07:19:03 RazaMac Installer[1417]:     -total-      27.18 seconds
Apr 24 07:19:03 RazaMac Installer[1417]: 
Apr 24 07:19:03 RazaMac Installer[1417]: IFDInstallController 153F4C0 state = 5
Apr 24 07:19:03 RazaMac Installer[1417]: Displaying 'Install Succeeded' UI.

```

chris1111, 2021-04-24:
Log is correct

RazaGR, 2021-04-24:
how can I debug this? I tried to install various times, It install the software but is unable to connect with adapter.