chris1111/Wireless-USB-Big-Sur-Adapter #105: After reboot Display Settings are broken

chris1111, 2021-06-17:
### Discussed in https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/discussions/104

<div type='discussions-op-text'>

<sup>Originally posted by **Pablow81** June 17, 2021</sup>
Hi there,

installing the "Wireless USB Big Sur Adapter-V7.pkg" by disabling SIP and allowing Gatekeeper worked fine.

But after reboot my display settings are broken, bad resolution that I cannot change and also a second display appeared on my Macbook Pro that doesn't exist in reality. The USB device seems to be detetcted now though.

With that resoulution and that settings I cannot work.

Then I have to reset PRAM and after that the Display Settings are fixed again but the extensions “Realtek Semiconductor Corp” and “MEDIATEK INC” are blocked (probably because SIP is now enabled again) and I can't use the USB device anymore.

Running on Big Sur 11.4.

Any suggestions? Thank you in advance.</div>

chris1111, 2021-06-17:
it seem you are using a patch for Old Mac ? is that the case you; should ask the Devs of this patch and ask why by installing two kext this happens.
I really don't understand how two simple kexts can broken display

Pablow81, 2021-06-17:
Apple MacBook Pro 15" Retina from Mid 2019.

chris1111, 2021-06-17:
This is Impossible

closed 2021-06-17 by chris1111