chris1111/Wireless-USB-Big-Sur-Adapter #119: No USB-WiFi Plugin

danielaconsti, 2021-06-30:
Hi, after successfully installing V7 as described in the instructions I'm still unable to use the adapter (RTL8188EU). 

[image]

I also did a fresh install of macOS Big Sur and reinstalled everything again and it doesn't work

[image]

Do you know what could be causing this issue?

chris1111, 2021-06-30:
RTL8188EU is not support anymore in new macOS  its old and is remove by realtek

closed 2021-06-30 by chris1111

danielaconsti, 2021-06-30:
Sorry, I have the RTL8812AU

It's an Alfa Awus036ach Wifi Dual Band clone

chris1111, 2021-06-30:
> Sorry, I have the RTL8812AU
> 
> It's an Alfa Awus036ach Wifi Dual Band clone

Is support by the program , make sur SIP is fully disable thens try other USB port

danielaconsti, 2021-06-30:
SIP is fully disabled and I've tried both USB ports :(

chris1111, 2021-06-30:
> SIP is fully disabled and I've tried both USB ports :(

Uninstall the program

then try this 

[Wireless USB Big Sur Adapter-V-7.zip](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/6744517/Wireless.USB.Big.Sur.Adapter-V-7.zip)

chris1111, 2021-06-30:
Uninstall the program but Reboot macOS 

then try this 

[Wireless USB Big Sur Adapter-V-7.zip](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/6744517/Wireless.USB.Big.Sur.Adapter-V-7.zip)

chris1111, 2021-06-30:
Uninstall like this
[image]

Then Reboot
after reboot use the program attaching UP

danielaconsti, 2021-07-01:
IT WORKS NOWWWW THANKS!

chris1111, 2021-07-01:
> IT WORKS NOWWWW THANKS!

Glad to hear that 😏

Luisalbert21, 2021-07-20:
[image]
Ya instalé correctamente la V7 pero mi antena mi macOs Big Sur sige sin reconocer la antena

Luisalbert21, 2021-07-20:
Le explico instalé el adaptador de usb para macOs Big Sur V7 en mi mac pero no detecta la antena si me puede ayudar porfavor