chris1111/Wireless-USB-Big-Sur-Adapter #126: Installation fails on macOS Big Sur 11.5

norbertschuler, 2021-07-22:
On macOS Big Sur 11.5 the installation fails as following (German version):

[image]

Are there any logs which I can check?

This is my csrutil status working well before:
```
$ csrutil status
System Integrity Protection status: unknown (Custom Configuration).

Configuration:
	Apple Internal: disabled
	Kext Signing: disabled
	Filesystem Protections: enabled
	Debugging Restrictions: enabled
	DTrace Restrictions: enabled
	NVRAM Protections: enabled
	BaseSystem Verification: enabled

This is an unsupported configuration, likely to break in the future and leave your machine in an unknown state.

```

norbertschuler, 2021-07-22:
My prior installed driver still worked after a restart of my MacBook.

closed 2021-07-22 by norbertschuler

chris1111, 2021-07-22:
This is strange I dont have issue on 11.5  update last night
[image]

**The Log can be save a the top bar - Window**
[image]

**Save**
[image]

**Show all Log**
[image]

norbertschuler, 2021-07-22:
I can share my install.log (this is from just running the installer again despite already having the software installed - I've not tried removing it before):

[install.log](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/6862000/install.log)

chris1111, 2021-07-22:
Missing Info from your LOG log is not complete
[image]

norbertschuler, 2021-07-22:
Sorry. Probably my `Filesystem Protections: enabled` made is fail as I did not disable everything with `csrutil` as you wrote in your description (s. above) - sorry for asking then:

[install.log](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/6863491/install.log)

chris1111, 2021-07-22:
Something wrong about your macOS  or the SIP ?
If you use a old macbook patching dosdude1 or OCLP 
try to disable SIP and File system 
csrutil disable
csrutil authenticated-root disable