chris1111/Wireless-USB-Big-Sur-Adapter #131: [SOLUTION] Big Sur 11.4

CaaioSB, 2021-08-06:
I saw many problems with the installation in Big Sur, I felt like sharing what I did to be able to use it, sometimes it can help someone and it doesn't cost me anything.

Firstly,

**Disconnect your device;**

**Disable SIP security:**

1. When connecting the device, go to the option "Recovery [...]";
2. On the recovery screen search for "Utilities" in the top menu and open the terminal;
3. Enter the following commands in the terminal:

```
csrutil disable
csrutil authenticated-root disable
reboot;
```

**Wait for your device to boot in normal mode**

1. Open Settings;
2. Go to Security & Privacy;
3. Unlock the padlock on the lower left side;
4. On the lower right side there will be a button (not Advanced), click on it and check "Realtek [...]", save and then click on "Allow", then restart your computer;

When your computer turns on again, just follow the steps that the application requires;
**If something** related to permissions appears, click on the button to go to "Security & Privacy" and allow again the update of "Realtek [...]"

**Put your wireless device and restart your computer again;**

While powering on, if your wireless device is flashing green, congratulations, you have wifi in your Big Sur. 🎊

closed 2021-08-06 by CaaioSB

chris1111, 2021-08-06:
thanks I know its also work with **csrutil authenticated-root disable**
I have already test

CaaioSB, 2021-08-06:
I'm the one who should thank you, thank you so much for this tool, you're amazing!

chris1111, 2021-08-14:
Did you try without csrutil authenticated-root disable  
latest Update Big Sur because a lot of Bug Fix on Big Sur 11.5.2 (20G95)

grimmwerks, 2021-09-15:
I am an idiot this absolutely fixed it for me.