chris1111/Wireless-USB-Big-Sur-Adapter #151: M1 chip

Gwindor, 2021-11-11:
There is a note that is doesn't work on M1. Do you think it is possible to make it work on it?

chris1111, 2021-11-11:
I dont think so

closed 2021-11-11 by chris1111

evg1n, 2021-12-05:
Hello,

I know this is closed but I wanted to share my experience.
My device is MBP 13" w/ M1. Below is my installer log. The AOT compiler fails when processing driver binaries. If you don't have an Apple Silicon, with your guidance, we can progress on my machine.

Appreciate your work. Thanks in advance.

`@(#)PROGRAM:Install  PROJECT:Install-1000
@(#)PROGRAM:Installer  PROJECT:Installer-1102
Hardware: MacBookPro17,1 @ 2.40 GHz (x 8), 16384 MB RAM
Running OS Build: macOS 12.0.1 (21A559)
Env: xfffff748aef51ad0=18446734490277845712 1
Env: xfffff74fb61edad0=18446734520462793424 1
Env: PWD=/private/var/folders/ts/hfqwnbt96jx0j0vlskbnt6mm0000gn/T/AppTranslocation/9E29B4E2-F748-41B2-AC6B-4F1CE8308B5D/d/Wireless USB Big Sur Adapter.app/Contents/Resources
Env: xfffff724b2c40ad0=18446734335722916560 1
Env: xfffff7369033ead0=18446734412452457168 1
Env: xfffff726655acad0=18446734343014107856 1
Env: ICNS=.
Env: xfffff72a5e51cad0=18446734360075946704 1
Env: xfffff74c5b9cfad0=18446734506059430608 1
Env: xfffff74ab6077ad0=18446734498986425040 1
Env: xfffff73ca0d70ad0=18446734438501386960 1
Env: xfffff73b81f30ad0=18446734433688160976 1
Env: xfffff74eaee07ad0=18446734516046297808 1
Env: xfffff7381f8c4ad0=18446734419152358096 1
Env: xfffff72d9bb15ad0=18446734373990521552 1
Env: xfffff7271399dad0=18446734345937476304 1
Env: xfffff72a37724ad0=18446734359423765200 1
Env: xfffff73c13fcfad0=18446734436138285776 1
Env: xfffff729f1904ad0=18446734358251326160 1
Env: xfffff737c4336ad0=18446734417619806928 1
Env: xfffff73dd1552ad0=18446734443609926352 1
Env: xfffff74a68ce7ad0=18446734497690843856 1
Env: xfffff724a239cad0=18446734335445420752 1
Env: xfffff73d870e2ad0=18446734442363759312 1
Env: USER=evg1n
Env: xfffff72a2bbdfad0=18446734359227398864 1
Env: xfffff73a4ac15ad0=18446734428467190480 1
Env: xfffff729fb547ad0=18446734358415178448 1
Env: xfffff73f260a6ad0=18446734449326058192 1
Env: xfffff74cd26f1ad0=18446734508052912848 1
Env: xfffff74f899e3ad0=18446734519716166352 1
Env: xfffff73c69d05ad0=18446734437578201808 1
Env: xfffff748ad837ad0=18446734490253621968 1
Env: xfffff725dd5faad0=18446734340732725968 1
Env: xfffff74ff91a6ad0=18446734521586576080 1
Env: COMMAND_MODE=unix2003
Env: xfffff73621fe8ad0=18446734410603465424 1
Env: xfffff74e11685ad0=18446734513404402384 1
Env: xfffff74efaedbad0=18446734517322234576 1
Env: xfffff749a9fc6ad0=18446734494489406160 1
Env: xfffff73de6f32ad0=18446734443972602576 1
Env: xfffff73c57033ad0=18446734437262768848 1
Env: xfffff74e68425ad0=18446734514861529808 1
Env: xfffff74e7cfa3ad0=18446734515209124560 1
Env: xfffff74c60f96ad0=18446734506149374672 1
Env: xfffff73de1990ad0=18446734443882810064 1
Env: __CFBundleIdentifier=com.apple.installer
Env: xfffff74827d5ead0=18446734488010877648 1
Env: LOGNAME=evg1n
Env: xfffff74e6a6e1ad0=18446734514897951440 1
Env: xfffff73b70510ad0=18446734433392331472 1
Env: xfffff72dfbb94ad0=18446734375601654480 1
Env: xfffff746c21eead0=18446734482009418448 1
Env: xfffff734f6d7aad0=18446734405584530128 1
Env: PATH=/usr/bin:/bin:/usr/sbin:/sbin
Env: xfffff73b127f7ad0=18446734431818316496 1
Env: xfffff74bbf6d9ad0=18446734503439080144 1
Env: xfffff7373d57fad0=18446734415357278928 1
Env: xfffff72865022ad0=18446734351598234320 1
[... 93 lines cut ...]
Using authorization level of root for IFPKInstallElement
Install request is requesting Rosetta translation.
Starting installation:
Configuring volume "Macintosh HD"
Preparing disk for local booted install.
Free space on "Macintosh HD": 151,31 GB (151306252288 bytes).
Create temporary directory "/var/folders/ts/hfqwnbt96jx0j0vlskbnt6mm0000gn/T//Install.2236HVY57b"
IFPKInstallElement (4 packages)
Current Path: /System/Library/CoreServices/Installer.app/Contents/MacOS/Installer
PackageKit: Enqueuing install with framework-specified quality of service (utility)
Running install actions
Removing temporary directory "/var/folders/ts/hfqwnbt96jx0j0vlskbnt6mm0000gn/T//Install.2236HVY57b"
Finalize disk "Macintosh HD"
Notifying system of updated components

**** Summary Information ****
  Operation      Elapsed time
-----------------------------
       disk      0.02 seconds
     script      0.00 seconds
       zero      0.01 seconds
    install      10.05 seconds
    -total-      10.07 seconds
IFDInstallController 16FE080 state = 5
Displaying 'Install Succeeded' UI.`

ezekiel747, 2022-03-02:
what would it take to support the M1 chip?
unfortunately for some of us, neither the official drivers are not supported (TP-Link in my case) and don't have any alternative.

chris1111, 2022-03-02:
You can try to use the [Notorized Guide](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/discussions/159) and 
csrutil enable --without kext
Not csrutil disable

shmizan, 2022-03-08:
@chris1111 thanks for all of your hard work.
all new macs are now Apple chips, and since TP-LINK and the rest of these companies can't seem to work out a solution, only you can help!
@evg1n has offered to help with your instructions on his machine.. maybe we can try that direction?
it will be very helpful since NO other solution exists.

jlk, 2022-06-16:
@chris1111 Just a data point - I tried 
```
csrutil disable
csrutil enable --witout kext
``` 
from recovery mode, but I still see "No USB-WiFi Plug in" from the menubar icon. This is an 16" M1 macbook pro.

I plugged the adapter into a raspi just to verify, and it's showing up as a realtek nic:
```
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xc811
  bcdDevice            2.00
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
  iSerial                 3 123456
```

Tobias-Conrad, 2022-07-21:
@chris1111 Chris what needs you to do it? And us to get Wifi working via your solution?