chris1111/Wireless-USB-Big-Sur-Adapter #156: No usb Wifi

sarrathy, 2021-11-30:
I tried using Archer T2U Nano . it still says no usb . Please help

chris1111, 2021-11-30:
Reboot then unplug adapter and replug

chris1111, 2021-11-30:
https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/discussions/83#discussioncomment-1722632

closed 2021-11-30 by chris1111

sarrathy, 2021-12-01:
> Reboot then unplug adapter and replug try diferant USB Port

Still no improvement , it still shows the same error .

sarrathy, 2021-12-01:
[image]

sarrathy, 2021-12-01:
[image]

sarrathy, 2021-12-01:
[image]