chris1111/Wireless-USB-Big-Sur-Adapter #165: X64 exception after installation

red5x, 2021-12-29:
Hi,

I reviewed instructions and videos.  I just installed this and that seemed to go fine.  On reboot I get an x64! exception after the Clover boot selector.  Tried booting to recovery and same result.  I had to boot to another drive and remove the 2 installed kexts from my efi manually.  After removal, booting is again possible.  

I'm running OS 10.15.7 on a i5 sandy bridge machine.  Legacy mode.  Using clover 5.21.0.0.  SIP was/is disabled.

Please advise if this is not compatible.

chris1111, 2021-12-29:
provide Debug LOG

red5x, 2021-12-30:
Hi,

Thanks for getting back to me.  My bootlog is attached.  Let me know what
you think.

Thank you  -

closed 2022-01-23 by chris1111