chris1111/Wireless-USB-Big-Sur-Adapter #168: Can not be used as part of Internet Shared

Johnne25, 2022-01-14:
You can not use this as a connection for the computer to use Mac's ethernet internet connection.

You can configure it, but it doesn't broadcast its SID so users can not connect to the Mac using WIFI.

chris1111, 2022-01-14:
Its WIFI not ethernet

[Read this](https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter/discussions/133)

Johnne25, 2022-01-15:
I understand that.  My Mac has an ethernet connection which is used for the Internet.
I want to use USB WIFI (and this driver) to setup Internet Sharing on the Mac.  Users connect to the Mac using WIFI to access its Ethernet internet connection.

chris1111, 2022-01-15:
Not possible to setup Internet Sharing with this Drivers

closed 2022-01-15 by chris1111