chris1111/Wireless-USB-Big-Sur-Adapter #176: source code / driver security

mtrezza, 2022-01-27:
Is the source code of the modified driver published anywhere? Or is there any other way to verify that the driver is save?

chris1111, 2022-01-27:
Package is not Sign, but its safe you can use the drivers and utility SIP enable
If you what to check it use Suspicious Package
[image]

closed 2022-01-27 by chris1111

chris1111, 2022-01-27:
If you whant to Sign in, Use CLI **productsign**
`productsign --sign "Developer ID Installer: Your Developer Name
(1A2B3C4D5E)" ~/Desktop/"USB Big Sur Adapter.pkg" ~/Desktop/signed-"USB Big Sur Adapter.pkg"`

mtrezza, 2022-01-27:
I think my question goes beyond signing, which is just a formal step. It would be interesting to know which changes have been made to the driver (if any) and which scripts are executed by the installer.

mtrezza, 2022-01-27:
I just found https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/discussions/159.

Are the drivers original Realtek drivers?

chris1111, 2022-01-27:
original