chris1111/Wireless-USB-Big-Sur-Adapter #179: Monterey 12.2.1

ZJSP, 2022-02-11:
Hiya, 
I have Moneterey 12.2.1 and a Archer T2U and for the love of me i cant get the software install be it this version of it or the officeal version which is onluy supported up to 10.1. 

is there anything i can do or is my virtual machine doomed to a ethernet connection forever. 

all the best 

ZJSP

chris1111, 2022-02-12:
Gives more infos about what append

ZJSP, 2022-02-12:
thank you for the quick reply mate. 

I'm running this on a 2019 Macbook Pro on Moneterey 12.2.1, running the V14 of your software.
I have disabled sip security and followed all of the instructions (i believe);
I open the installer and it gets to about 85% throught the installation then it stops on "Running Scripts"  and hits me with a "The Installation failed" "The Installer encounted an error that caused the installation to fail. Contact the software manufacturer" 

I've tried pretty much everything i can think of and it wont have it. 

any help or suggestions massivly appreciated. 

ZJSP

chris1111, 2022-02-13:
If the SIP is disable the installer not suppose to be failled
issue is from your macBook trust me

closed 2022-02-13 by chris1111