chris1111/Wireless-USB-Big-Sur-Adapter #213: Change speedtest.py to Ignore Cert Please

DennisFaucher, 2022-06-14:
Thank you

/Private/tmp/speedtest.py
Retrieving speedtest.net configuration...
Cannot retrieve speedtest configuration
ERROR: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>

chris1111, 2022-06-14:
Python war remove from macOS since Monterey 12.2
If you have python3 installed in macOS Monterey
use this app

[Network Speed Test-V3.zip](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/8903703/Network.Speed.Test-V3.zip)

DennisFaucher, 2022-06-14:
[image]

DennisFaucher, 2022-06-14:
Thank you. That fixed it.

✦ 🕙 19:19:43 ✅  cd /Applications/Python\ 3.10/

✦ 🕙 19:20:02 ✅  ./Install\ Certificates.command
 -- pip install --upgrade certifi
Collecting certifi
  Downloading certifi-2022.5.18.1-py3-none-any.whl (155 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 155.2/155.2 kB 403.9 kB/s eta 0:00:00
Installing collected packages: certifi
Successfully installed certifi-2022.5.18.1
 -- removing any existing file or link
 -- creating symlink to certifi certificate bundle
 -- setting permissions
 -- update complete

chris1111, 2022-06-14:
Now you can use  Network Speed Test-V3 😇