chris1111/Wireless-USB-Big-Sur-Adapter #237: TP-Link AC1300

DanielRomanMartinez, 2022-10-20:
As Chris1111 said, my previous "post" / "thread" in discussions is a issue.

https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/discussions/236#discussioncomment-3927926

So, i think that must be here, if I am wrong just tell me. Sorry for all problems i can cause.

**Specs:**

- OS: Mac OS Monterey 12.6
- Computer: MacBook Pro (16-inch, 2019)

I've seen also the the video that Chris posted (https://www.youtube.com/watch?v=YqZAy8jntow) 

And I think that i didn't miss a step.

chris1111, 2022-10-20:
I have the same adapter and its work from 10.15 to macOS 13 without any issue

DanielRomanMartinez, 2022-10-20:
Hello chris, I 've seen your previous answer. And i couldn't contest.

It's possible that i am missing some command in terminal or disable something? 

Can I execute some command and help you to figure out what is happening?

DanielRomanMartinez, 2022-10-20:
Finally seems like the network must be configured manually. DHCP doesn't work properly.

Also I think that public networks that requires to accept some permissions won't work properly. I will keep investigating, and i will attach here the results.

I think that i can close it anyway, are you agree ? @chris1111

closed 2022-10-20 by chris1111