chris1111/Wireless-USB-Big-Sur-Adapter #240: crashing

Gizmo12344, 2022-10-22:
The Big Sur would crash my entire MacBook. Several weeks before it was working perfectly, until now. It would randomly crash my entire computer whenever I would enter the adapter, Why is that? First I would connect it and then the adapter would blink super fast rather than slowly. Then It would cause the entire thing to just freeze after 5 minutes of use. Please help, thank you for your consideration.

chris1111, 2022-10-22:
The adapter as nothing to do with a crashing or anything else issue of crashing is on you side

closed 2022-10-22 by chris1111

lycaeides, 2024-04-12:
> The Big Sur would crash my entire MacBook. Several weeks before it was working perfectly, until now. It would randomly crash my entire computer whenever I would enter the adapter, Why is that? First [...]

Did you solve it ?