chris1111/Wireless-USB-Big-Sur-Adapter #242: can't execute the command /usr/sbin/spctl kext-consent add ZYM2ETK3E7 says spetl: failed to store new configuration.

HDV998, 2022-10-25:
hi, I had version 13 on Monterey and It worked great until I upgraded to Ventura, It said that the app was damaged and moved It to the bin. then I downloaded the version 14, disabled SIP, installed it and then when I reenabled SIP, it caused the same issue. then I realised that I forgot to run the command "/usr/sbin/spctl kext-consent add ZYM2ETK3E7" so I disabled SIP, uninstalled it and ran the terminal command but it gave this error "spetl: failed to store new configuration." what to do, please help

HDV998, 2022-10-25:
Also I am using opencore 0.8.5

chris1111, 2022-10-25:
spctl kext-consent disable
spctl kext-consent list

closed 2022-10-25 by chris1111

HDV998, 2022-10-26:
[image]
still doesn't work, same error when writing the spctl kext-consent disable command as well