chris1111/Wireless-USB-Big-Sur-Adapter #250: No USB Plug in

PK268, 2022-12-01:
Hey, sorry for this because this is probably like the thousandth no usb plug in issue you've seen but I have an ALFA AWUS1900 and i can't get the adapter to work. Here's what I've already done:

Disable SIP
Uninstall and then reinstall adapter
Different USB port

chris1111, 2022-12-01:
what machine are you trying this on?

PK268, 2022-12-01:
Macbook pro 2019 16" intel i9

chris1111, 2022-12-01:
> Macbook pro 2019 16" intel i9

Try this please [Wireless USB Big Sur Adapter-V15-1.zip](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/files/10135775/Wireless.USB.Big.Sur.Adapter-V15-1.zip)

PK268, 2022-12-02:
Sorry, it still says no USB plug in

chris1111, 2022-12-02:
So its not support for macOS 11,12,13

closed 2022-12-02 by chris1111