chris1111/Wireless-USB-Big-Sur-Adapter #261: With SIP enable

Rxxbertx, 2023-02-24:
I've been doing the same process 5 times and it doesn't work with SIP enable OC hackintosh

chris1111, 2023-02-24:
you did something wrong because it works, I have tested everything from Big Sur 11 to Ventura 13

Rxxbertx, 2023-02-24:
I have carried out the phase of having the SPI active, then execute Consent.command, reboot in recovery And put SIP disable and /use.....kext-consent add.... I have rebooted, I have installed the wifi adapter I opened the config.plist file to make an OC clean Snapshot, I have rebooted, I have connected to the wifi network, I have rebooted in recovery and put SIP ENABLE, I have rebooted and the app disappears saying That there has been a problem with the app :/

Rxxbertx, 2023-02-24:
MacOS: Ventura 13 with OPENCORE

Rxxbertx, 2023-03-01:
Okey, i found the problem... U need to open settings , privacy. And allows statusappbar.... (I have disable spctl.. and still ask for permission to open it)

closed 2023-03-01 by chris1111