chris1111/Wireless-USB-Big-Sur-Adapter #266: MacOS Ventura complains the StatusBarApp.app is damaged, thus cannot open

cc2lee, 2023-04-12:
After successful installation on MacOS 13 (Ventura), root the Mac and the MacOS pops up a dialog, saying the StatusBarApp.app is damaged and needs to be moved into Trash Bin.

b0bd, 2023-04-15:
same issue on 13.3.1.   Using OpenCore 0.9.1 and the OC installer.   StatusBarApp.app is damaged, unless csrutil disable is run.

It did work on Monterey with the kext-consent command, but that doesn't work on Ventura.

chris1111, 2023-04-15:
> same issue on 13.3.1. Using OpenCore 0.9.1 and the OC installer. StatusBarApp.app is damaged, unless csrutil disable is run.
> 
> It did work on Monterey with the kext-consent command, but that doesn't work on Ventura.

Of course not working with OCLP
This program work perfectly if you know what you did even with OCLP

closed 2023-04-15 by chris1111