chris1111/Wireless-USB-Big-Sur-Adapter #282: Doesn't work with Realtek 0xf179 on Ventura

egslava, 2023-08-05:
Hello! Thank you for doing this amazing stuff! I didn't believe I would be able to find these drivers, to be honest (◕‿◕)

Today I bought realtek 0xf179 wi-fi adapter and tried to make it work (`csrutil disable` in safe mode and, later, in normal mode, sudo spctl --master-disable)  on macos ventura (13.4 (22F66), mbp 2018), but it says, that there's no wi-fi adapter.

Specifications:

```
>         802.11n:
> 
>           Product ID: 0xf179
>           Vendor ID: 0x0bda  (Realtek Semiconductor Corp.)
>           Version: 0.00
>           Serial Number: 90DE808DCCA9
>           Speed: Up to 480 Mb/s
>           Manufacturer: Realtek
>           Location ID: 0x14100000 / 7
>           Current Available (mA): 500
>           Current Required (mA): 500
>           Extra Operating Current (mA): 0
```

What could go wrong?

[image]

chris1111, 2023-08-05:
What brand and model of USB adapter are you using? Without that I can't help you

egslava, 2023-08-08:
Thank you for your reply. What do you mean by brand and model? I thought, "realtek 0xf179" would be enough, but I'm not familiar much with chips/electric engineering or drivers.

Here I found the photo of the package in the internet:
[image]

So the brand is "i-rock". When I tried to find 'realtek 0xf179 brand name', I found "RTL8188FTV" in this issue ([102](https://github.com/chris1111/Wireless-USB-Adapter/issues/102)) in this repo, lol. Basically, it ends up, that this wi-fi adapter is not supported (◕‿◕)

Anyways, thank you for your time!

chris1111, 2023-08-08:
Exactly not supported

closed 2023-08-08 by chris1111