chris1111/Wireless-USB-Big-Sur-Adapter #291: USB Adapter not active after Big Sur installation

chris1111, 2023-09-21:
### Discussed in https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/discussions/289

<div type='discussions-op-text'>

<sup>Originally posted by **Augeydoggy** September 20, 2023</sup>
I installed the Big Sur drivers on my 2015 Macbook Pro running Monterey 12.6. I disabled the SIP, gave the necessary permissions, and it said it had installed. I am using a Netgear A6150, which now shows up in the network preferences, but as not connected. I am not a programmer, but all the steps seemed to have gone correctly, so don't know what to do next. </div>

Augeydoggy, 2023-09-22:
Somehow after the most recent Monterey update the A6150 is now connected. However I still cannot get the wifi turned on:
[image]

Augeydoggy, 2023-09-22:
After updating to Monterey 12.7, the A6150 suddenly connected. But I still cannot get the WiFi started:

[image]

chris1111, 2023-09-22:
[image]

closed 2023-09-22 by chris1111

Augeydoggy, 2023-09-22:
> <chris1111/Wireless-USB-Big-Sur-Adapter/issue/291/issue_event/10450740287@
> github.com>

Doh! It's working now, tho it took a few tries to actually connect to our
network. THANKS!