chris1111/Wireless-USB-Big-Sur-Adapter #292: MacOS Sonoma - instant crash connecting to network

elixir-of-life, 2023-10-12:
Hi there, wondering if any one can give me some pointers on this issue?

I followed the videos/instructions on two almost identical Macbook Pro 16 (2020) with Intel chipset.  Both systems run MacOS Sonoma 14.0 and are fully patched.

I have the Archer T4U Plus, and the driver certainly is communicating with the card as I manage to scan for networks fine. I can also disable/re-enable the card.

I have two wifi networks in my house. One of them works correctly and the other network gives me instant crashing.

On the one that crashed - when I go to connect to my network, the instant I put the password in I get an instant crash and reboot. I noticed if I mistype the wifi password it fails correctly, but when the password is correct - instant crash.

I tried this with csrutil fully disabled. I tried with adding the csrutil kext consent also and enabling csrutil

I had the same result on both machines for everything I tried. 

Does the application provide logs?  Where can I find these?

chris1111, 2023-10-12:
For Logs check in the console
Your problem seems quite complex, you use OCLP and you say you have two Networks and one which crashed you also say you used the kext consent drivers but it seems impossible because your Sonoma system would not start with the SIP activated so it is hard to follow

elixir-of-life, 2023-10-16:
I don't think its complex at all.  I run standard Macs with Intel. I don't know what OCLP is but i am using the stock OS.  By fully patched i mean that all updates are the latest, I am just using terminology I use daily as I work on application security.

The Sonoma system started with and without csrutil enabled. Back to my original question - does the application provide logs?  Thanks.

closed 2023-10-26 by chris1111