chris1111/Wireless-USB-Big-Sur-Adapter #294: Realtek USB WiFi adapter not recognized

m0reilly, 2023-10-25:
T3U Plus adapter not recognized as connected in Network settings.
[image]

chris1111, 2023-10-26:
You don't give enough information
what machine to use, what version of macOS

m0reilly, 2023-10-26:
I'm sorry, a Ryzentosh running Sonoma 14.1 and using the  Archer T3U Plus.
I followed your instructions, turned SIP off, ran the consent command,
reset SIIP,  and rebooted.  Sonoma states that the dongle is not connected.
This is what is displayed  in system network, but it is described as
'ethernet'...

Michael

m0reilly, 2023-10-26:
[image: Screenshot 2023-10-26 at 6.30.53 AM.png]

chris1111, 2023-10-26:
If you use Hackintosh why using this program better to using the one for OpenCore 
So uninstall this one  reboot then try the OpenCore version following the guide

m0reilly, 2023-10-26:
Which one is for opencore? So your guide is only for real macs? I was told
this works on hackintoshes...are you referring to OCLP?

m0reilly, 2023-10-26:
Welcome to Wireless USB OC Big Sur Adapter
<https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter#welcome-to-wireless-usb-oc-big-sur-adapter>Drivers
for Realtek 802.11n and 802.11ac USB Wi-Fi adapters
<https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter#drivers-for-realtek-80211n-and-80211ac-usb-wi-fi-adapters>Support
macOS Sonoma 14 - macOS Ventura 13 - macOS Monterey 12 - macOS Big Sur 11 -
macOS Catalina 10.15
<https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter#support-macos-sonoma-14---macos-ventura-13---macos-monterey-12---macos-big-sur-11---macos-catalina-1015>

   - *For Hackintosh PC and Laptop only*

chris1111, 2023-10-26:
Read the pages

m0reilly, 2023-10-26:
I read what was available to me. I followed the instructions. I am not sure
what other pages you are referring to. Others are having the same issue, so
it would be neat if there was a general layout of the steps you took other
than what you have posted. Could the AMD patches be an issue for me? I'll
see if I can find other Ryzentosh users that have had success...

Thanks again,
Michael

chris1111, 2023-10-27:
The page is verry clear this is what is write here ⬇︎

[image]

closed 2023-10-27 by chris1111

m0reilly, 2023-10-27:
That is the procedure and drivers that I used, I even tried it 3 times and
still the dongle is not recognized....