chris1111/Wireless-USB-Big-Sur-Adapter #298: System Extension blocked

filipopo, 2023-12-22:
I'm using MacOS 14.1.2 on a M1 pro macbook and I got this message during install
[image]
....... I just read that it doesn't work on Apple Silicone .... is this the reason the system extensions are blocked?

And is there something we can do to make it run on M1, M2, M3, ... maybe even file a bug report to apple? I don't know which USB Wifi dongle to get if this doesn't work on M1 Mac...

chris1111, 2023-12-22:
It does not work on Apple Chip

closed 2023-12-22 by chris1111

filipopo, 2023-12-22:
You are very helpful, maybe the problem is not in the apple chip and also your repo will die out because no new hardware is made without an apple chip

chris1111, 2023-12-22:
> You are very helpful, maybe the problem is not in the apple chip and also your repo will die out because no new hardware is made on intel chips

You are Wrong ☛ [OCLP](https://github.com/dortania/OpenCore-Legacy-Patcher/)

filipopo, 2023-12-23:
So you will forever use 2007 hardware? okay man whatever