chris1111/Wireless-USB-Big-Sur-Adapter #305: No USB-WiFi plug in

beckzhao0410, 2024-02-25:
macOS Monterey 12.6.9
Already disable SIP
Installed the Wireless USB OC Big Sur Adapter-V16 
Plug in my USB wifi, it is comfast CF-811AC V3 
Opencore config as blow:
[image]
But after I restart, usb wifi cannot use and hint "No USB-WiFi plug in"
[image]
I don't know why has this issue

chris1111, 2024-02-25:
Did you try differant USB port?

beckzhao0410, 2024-02-26:
Yes, I tried all USB ports.
When I use "kextstat | grep -v "RtWlan" in Terminal, there is nothing. So, I think the kexts are not loaded successful.

chris1111, 2024-02-26:
If is not load succesful from OC its probably a config issue? , check correctly

beckzhao0410, 2024-02-27:
I am not sure, would it conflict with other config?

beckzhao0410, 2024-03-06:
I changed another USB WiFi, this issue resolved

closed 2024-03-06 by beckzhao0410