chris1111/Wireless-USB-Big-Sur-Adapter #306: No USB-WiFi plug in - ASUS adapter

romuloslv, 2024-02-26:
MacOS: Sonoma 14.2.1 (23C71)
Model Name: MacBook Pro
Model Identifier: MacBookPro15,1
Processor Name: 6-Core Intel Core i7
USB-C Mobile Adapter: DELL DA310u
Wifi Adapter: ASUS USB N-10
SIP: disabled

Tested the adapter wifi on a Windows 11 notebook (working)
Tested the two usb ports of the DELL usb-c adapter with different usb devices (working)
Tested the adapter wifi on the two existing USB ports of the DELL usb-c adapter (not working)

He is even recognized by the OS, but does not show any signs of life.
Can you help me?

[image]

[image]

chris1111, 2024-02-26:
Asus N10 is not in the support list then if I remember this adapter cause trouble for many years !

when iam back home tonight I wil investigate

romuloslv, 2024-02-26:
What support list?
I found these records in the [README](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter?tab=readme-ov-file#support-drivers-adapter):

ASUS_USB-N10E_92CU
ASUS_USB-N10_92CU
ASUS_USB-N10E_92CU
ASUS_USB-N10_92CU

thanks in advance.

chris1111, 2024-02-26:
Yea this is old probably 10 years this list is from the driver but in the pass 
See Know working testing adapter 
That’s why i always update the working list

romuloslv, 2024-02-26:
I understand, I will wait for you to debug at night. Maybe you'll find something that can make it work.
It's really old, bought it in 2012. To my surprise, I found it thrown in the corner of the drawer, that's when I decided to try to make it work.

chris1111, 2024-02-27:
I looked for your N10 and if it doesn't work I can't do anything because I told you it was a problematic adapter in macOS and it hasn't improved in all this time. Sorry

chris1111, 2024-02-27:
You can try this terminal command

```
cd /Library/Extensions/
sudo chown -R root:wheel RtWlanU1827.kext
sudo chown -R root:wheel RtWlanU.kext

```

In the second command maybe yousee this like me just trying

Click on OK to the PopUp
[image]

Continue the third command

[image]

then go here then Allow

[image]

romuloslv, 2024-02-27:
Followed your tutorial, but still no sign of life from the adapter.

1. Uninstallation (by Wireless USB Big Sur Adapter);
2. Restart;
3. Set permissions
4. Installation (by Wireless USB Big Sur Adapter);
5. Restart

Also tried changing the order from 3 to 4 in a new installation, without success.
Checked the permissions, as well as the driver within the OS and nothing. Everything is there and looks fine, but nothing is working.

I know you've already wasted some time on this (no guarantees), but is there any debug or change we can make to force these kexts to load and make it work?

[image]

chris1111, 2024-02-27:
Your just reinstall OK
Try the command propose upper

romuloslv, 2024-02-29:
I redid it, but nothing yeat. Bought another adapter, I'm waiting for it to arrive so I can continue testing and update this thread.

romuloslv, 2024-03-11:
Updating

The new USBC -> USB adapter has arrived, tested on Windows 11/Sonoma 14.

Again, the usb wifi adapter only works in Windows, in Sonoma the hardware can be read just like we saw in the old DELL adapter, but the driver is not loaded in the OS. So the Asus dongle seems to be sleeping forever.

Just to confirm, this is model "ASUS USB-N10 ([ASUS EZ N](https://www.amazon.com/dp/B003E6493G))" tested by you that apparently worked?
I say this because I know of the existence of a model "ASUS USB-N10 ([NANO B1](https://www.asus.com/networking-iot-servers/adapters/all-series/usb-n10-nano-b1/))" with similar nomenclature.

chris1111, 2024-03-12:
Test by me? I never use Asus adapter then I told you its a problematic adapter 
Your ASUS EZN on amazon show working in macOS 10.6 
Why you not buying a Testing and working adapter

closed 2024-03-12 by chris1111

romuloslv, 2024-03-12:
Thought that in this [test](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/issues/306#issuecomment-1965717943) you were using the Asus USB N10, no problems.
I'm going to buy a new usb wifi adapter that has support, I just tried to use something that I've had for a while and that I believed was supported with its installation package.

Thank you for your help